bigtreetech/BIGTREETECH-SKR-V1.3 #181: skr 1.4 x and y min endstop wont trigger

nevertyam, 2019-12-30:
i havw the ground to signal wires connected butcant get a trigger when i do a m119

nevertyam, 2019-12-30:
cutoff the pin for tmc 2209 if not using stallguard fixed this...

nevertyam, 2020-01-07:
I was using 2209 driver in uart and without stall guard. So the y and x weren’t triggering what so ever. The fix was to cutoff that diag pin just like the instructions

Gerggg, 2020-01-07:
Cutting a pin does not seem to me to be a good solution ... especially with a SKR 1.3 I do not have the problem.
I'm going to look for a less definitive solution.
thank you

Gerggg, 2020-01-07:
I'm going to see if on the Marlin side, they manage to find a sofware solution

nevertyam, 2020-01-07:
Yeah I would agree. They made the 1.4 and 1.3 issues on the same github. But the title says 1.4 and so this fix is for the 1.4 board with tmc2209 only. So it’s pretty specific.

For you however, do a m119 command. If it’s already triggered without physically being triggered, that’s an easy fix, just invert it in firmware.

Gerggg, 2020-01-07:
I do not see how, what is the option in Marlin?

nevertyam, 2020-01-07:
do a m119 command. If it’s already triggered without physically being triggered, that’s an easy fix, just invert it in firmware. 

Look for X_MIN_ENDSTOP_INVERTING false
And set to true IF your getting a triggered command on a m119

Toby00, 2020-01-11:
I have the same problem whit tmc2209 i have cut the pin off.
The endstops are not working i have tmc 2209 v1.2

lijinlin19970813, 2020-01-14:
> I was using 2209 driver in uart and without stall guard. So the y and x weren’t triggering what so ever. The fix was to cutoff that diag pin just like the instructions

In fact, it is not necessary to cut the pin, you only need to bend the pin so that it does not insert into the hole.

c3D-Dan, 2020-05-15:
Sounds like your signal voltage is too high.  Please check your 3.3v Rail is actually measuring 3.3v .   If it's measuring higher, replace your AM1117 LDO Regulator

nicobits46, 2020-09-15:
hi i´m having the same problem with a skr 1.4 turbo.
x and y won´t home, i cut off the diag pin because i want to use the endstops but i can{t get them to home.
the endstops work perfect doing an M119 . it´s making me crazy thank you!!!

rkarlsba, 2021-02-04:
This is the first time I've read a manual that tells me to destroy my hardware to fix it. Perhaps it should be better to fix this properly?

c3D-Dan, 2021-02-04:
You should have to.  The SKR boards are poorly designed. Replace it with a GTR.