bigtreetech/BIGTREETECH-SKR-V1.3 #184: SKR v1.4 - Schematic diagram is missing

bianchifan, 2019-12-31:
see title

GadgetAngel, 2020-01-25:
Where us the Schematic diagram for the SKRV1.4.  You can not expect us to believe that the LPC1768 chip and the LPC1769 chip are the same.  I really would like to see where each of the pins are coming from the LPC1769 chip. Please publish your schematic diagram for LPC1769 (SKR V1.4 Turbo) like you do for the LPC1768 (SKR V1.3)

MowMoney, 2020-01-25:
> Where us the Schematic diagram for the SKRV1.4. You can not expect us to believe that the LPC1768 chip and the LPC1769 chip are the same. I really would like to see where each of the pins are coming [...]

NXP Semiconductor, the people that make the chip, show the pin layout for the chip is identical. They have a very detailed pdf available if you care to research this for yourself.

GadgetAngel, 2020-01-26:
> NXP Semiconductor, the people that make the chip, show the pin layout for the chip is identical. They have a very detailed pdf available if you care to research this for yourself.

OK, if that is the case then why do I see differences in the pins files.  For example here is my screen shot.  The file opened on the LEFT is the SKRV1.4 pins file, the file opened on the RIGHT is the SKRV1.3 pins file.   THIS is why I want to see the schematic.  If the pinout on each chip is exactly the same then why the different pin numbers on these items?  I just downloaded the Marilin-bugfix-2.0.x today.  If you could help me understand the difference I would be very grateful
[image]

MowMoney, 2020-01-26:
> OK, if that is the case then why do I see differences in the pins files. For example here is my screen shot. The file opened on the LEFT is the SKRV1.4 pins file, the file opened on the RIGHT is the [...]

I see your point. And I honestly have no answer for this. I am no expert on this subject by any stretch. I had simply looked into the chip as I was curious about the differences, which I understood to be clock rate. 

My guess is the chips have the same basic pin layout and the boards are designed differently. I don’t believe the chip cares what pin does what, but more the internal geometry of the board dictates which pins are assigned to what function.

michielbaird, 2020-01-29:
I am very interested on this. The schematic is the best way to see if there are adequate, protections build into the circuit and how it can interact with external components, a pinout diagram is not sufficient. For example what is the size of the reference resistors used on the ADCs, are there additional ADC pins available etc.

DeviousPenguin, 2020-01-29:
I'd be interested in a schematic also, as I may look at using this board as a part of a 36v setup.

michielbaird, 2020-03-03:
On the Facebook group I've now seen several boards that have burned. It's difficult to diagnose these issues without having access to the schematic. Im assuming there is a reason this is being withheld.

asdmin, 2020-03-09:
dup to https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/184

DeviousPenguin, 2020-03-12:
> On the Facebook group I've now seen several boards that have burned. It's difficult to diagnose these issues without having access to the schematic. Im assuming there is a reason this is being withheld.

It seems the SKR V1.3 can be run with 36V for the motors if the 35V caps are replaced, but for now I'm sticking with 24v as I don't want to burn my board.

Do you have a link to the burned boards, I'm interested in seeing what *not* to do! :)

closed 2020-03-12 by bianchifan

laserbudd, 2020-03-14:
Hi, I am really a nooby in this 3D printing, but I have a question about this board (SKR 1.4 Turbo).
I have a modded Anet A6 (A8, supposedly) that has a Anet V1 board on it.  I am wanting to upgrade it to the SKR 1.4, and be able to use it for learning on, as I am eventually want to build a "cube" printer.
Can anyone possibly give me a wiring diagram so I can do this?
Thanks for any help I can get and, if I am making an error posting in this thread, would a Moderator please tell me?