bigtreetech/BIGTREETECH-SKR-V1.3 #185: SKR 1.3 BLTouch Bilinear Bed leveling Marlin 2.0

rainerschulte, 2019-12-31:
Hi folks!
I have my Anet PK8 equipped with SKR1.3 and TFT35, newly added a BLTouch. 
When running G28 the Nozzle is not hitting the Middle of the Board and G29 starts Leveling outside left of the Heatbed. 
Therefor I have changed the Zero-Point of the Nozzle from the Corner Left down around 30 mm to the right, so that the BLTouch is now hitting the former 0-Point off the Nozzle. 
Now the Nozzle runs out on the right while the BLTouch is still on bed...

`#define NOZZLE_TO_PROBE_OFFSET { -30, -3, 0 }`
`#define MIN_PROBE_EDGE 15`
`// The size of the print bed
#define X_BED_SIZE 215
#define Y_BED_SIZE 215

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS -58    // normally -28, but added Nozzle offset!
#define Y_MIN_POS -23
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 230`

The Offset from Nozzle to BLTOUCH seems beeing ignored. 
Safe Homing is activated.

Ferrari2, 2020-01-26:
Hi
I have the exact same problem using G29 for leveling - the nozzle runs outside the heat bed...
Do you know any solution.
Kind regards
Bjarke

rainerschulte, 2020-01-26:
Hi Ferrari,
yesterday I found out the Solution. The holy secret is in configuration.adv.h!
Since I have the Code on my other PC I will post the Lines accordingly tomorrow.

Ferrari2, 2020-01-26:
I have problems with the connected LCD too. Tomorrow I'm going to test the two flat cables :-)

rainerschulte, 2020-01-27:
Hi Bjarke,

first you  should check the following lines for the LCD. I'm using the TFT35 Controller with the knob to switch between the Marlin-Mode and Touchscreen.

```

//RepRap_Discount FULL GRAPHIC Smart Controller
// http://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller
//
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
```
Of course the wiring should be correct..

In my particular case I've converted an ANET PK8 to "Bigtreetech PK8" ;-)
So the Z-Probe and BLTOUCH should look like this in Configuration.h:
```
/**
 * Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
 *
 * Enable this option for a probe connected to the Z Min endstop pin.
 */
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

/**
 * The BLTouch probe uses a Hall effect sensor and emulates a servo.
 */
#define BLTOUCH

 * Specify a Probe position as { X, Y, Z }
 */
#define NOZZLE_TO_PROBE_OFFSET { -30, -3, 0 }  // This is to be measured out depending on the Placement of your Probe!!!

// Certain types of probes need to stay away from edges
#define MIN_PROBE_EDGE 10

// The size of the print bed (in this case I reduced it because of Glas)
#define X_BED_SIZE 215
#define Y_BED_SIZE 215

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS -28
#define Y_MIN_POS -24
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 220

//#define AUTO_BED_LEVELING_3POINT
//#define AUTO_BED_LEVELING_LINEAR
#define AUTO_BED_LEVELING_BILINEAR
//#define AUTO_BED_LEVELING_UBL
//#define MESH_BED_LEVELING

/**
 * Normally G28 leaves leveling disabled on completion. Enable
 * this option to have G28 restore the prior leveling state.
 */
#define RESTORE_LEVELING_AFTER_G28

#define Z_SAFE_HOMING

#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT ((X_BED_SIZE) / 2)    // X point for Z homing when homing all axes (G28).
  #define Z_SAFE_HOMING_Y_POINT ((Y_BED_SIZE) / 2)    // Y point for Z homing when homing all axes (G28).
#endif
```
By searching the required Lines in Editor you can make the above mentioned Settings for your own. TIPP!! Comment out default Settings, copy them a line below and remove Comments (//) for your own settings. So you can easily return to the Beginning...
Somewhere at the End of the Script you will find additional Settings for BLTouch, if you use this... 
```

//regarding BLTouch...
/**
 * Number of servos
 *
 * For some servo-related options NUM_SERVOS will be set automatically.
 * Set this manually if there are extra servos needing manual control.
 * Leave undefined or set to 0 to entirely disable the servo subsystem.
 */
#define NUM_SERVOS 1 // Servo index starts with 0 for M280 command

// (ms) Delay  before the next move will start, to give the servo time to reach its target angle.
// 300ms is a good value but you can try less delay.
// If the servo can't reach the requested position, increase it.
#define SERVO_DELAY { 300 }
```

OK... Let's get to the real Magic.. the **Configuration.adv.h**

```
#if PROBE_SELECTED && !IS_KINEMATIC
  //#define MIN_PROBE_EDGE_LEFT MIN_PROBE_EDGE
  #define MIN_PROBE_EDGE_RIGHT MIN_PROBE_EDGE
  #define MIN_PROBE_EDGE_FRONT MIN_PROBE_EDGE
  #define MIN_PROBE_EDGE_BACK MIN_PROBE_EDGE
  #define MIN_PROBE_EDGE_LEFT 40
  //#define MIN_PROBE_EDGE_RIGHT 180
  #define MIN_PROBE_EDGE_FRONT 15
  //#define MIN_PROBE_EDGE_BACK 185
#endif
```
Here you will find my Idea behind commenting out Defaults and personal Settings. These were formerly 4 Lines, ending on _MIN_PROBE_EDGE_

I added my hardcoded Positions, I figured out by moving the x and y-axis to the best Probing points. You should figure out your own values. Comiling shows Errors regarding oundaries and failed, so I could set the Values accordingly.

A good source to get Answers to SKR1.3 and additional Stuff is this american guy:
[(https://www.youtube.com/channel/UCqRiv7rQuxge63bqJ2hVNUQ)]
Listen carefully (also between the lines...) to him got me to find my Solutions for my Problems.

closed 2020-01-27 by rainerschulte