bigtreetech/BIGTREETECH-SKR-V1.3 #186: Marlin 2.0 bugfix for SKR v1.3

dabossie, 2020-01-02:
Hello all,
I'm playing around with an SKR V 1.3 for my ANET AM8.
I use the Marlin 2.0.x bug fix and get 5 errors during compiling.
I'm using VSC and Atom as build platform.
[image]

**Is there somebody who can tell me what I did wrong**
**To the best of my knowledge I did all there is to do.**

After the compile VSC terminal window is showing:

> Executing task: C:\Users\HP\.platformio\penv\Scripts\platformio.exe run <

Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.0.zip; framework: arduino; board: nxp_lpc1768)
-------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x 0.1.0 > NXP LPC1768
HARDWARE: LPC1768 100MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES: toolchain-gccarmnoneeabi 1.80201.190214 (8.2.1), framework-arduino-lpc176x 0.2.2
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 6 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <LiquidCrystal> 1.0.0
|-- <U8glib-HAL> 0.4.1
|-- <TMCStepper> 0.5.2
|-- <Adafruit NeoPixel> 1.3.0
|-- <SailfishLCD>
Building in release mode

Unable to find destination disk (Autodetect Error)
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) or copy the firmware (.pio/build/LPC1768/firmware.bin) manually to the appropriate disk

Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\DebugMonitor.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL_SPI.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\MarlinSerial.cpp.o
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfigPre.h:37,
                 from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\HAL_LPC1768\HAL.cpp:25:
Marlin\src\HAL\HAL_LPC1768\../../inc/../../Configuration.h:133: warning: "BOARD_BIGTREE_SKR_V1_3" redefined
  #define BOARD_BIGTREE_SKR_V1_3

In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfigPre.h:35,
                 from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\HAL_LPC1768\HAL.cpp:25:
Marlin\src\HAL\HAL_LPC1768\../../inc/../core/boards.h:213: note: this is the location of the previous definition
 #define BOARD_BIGTREE_SKR_V1_3        2016  // BIGTREE SKR_V1.3 (Power outputs: Hotend0, Hotend1, Fan, Bed)

In file included from Marlin\src\HAL\HAL_LPC1768\../../core/../inc/MarlinConfigPre.h:37,
                 from Marlin\src\HAL\HAL_LPC1768\../../core/../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\HAL_LPC1768\../../core/serial.h:24,
                 from Marlin\src\HAL\HAL_LPC1768\DebugMonitor.cpp:26:
Marlin\src\HAL\HAL_LPC1768\../../core/../inc/../../Configuration.h:133: warning: "BOARD_BIGTREE_SKR_V1_3" redefined      
  #define BOARD_BIGTREE_SKR_V1_3
 
In file included from Marlin\src\HAL\HAL_LPC1768\../../core/../inc/MarlinConfigPre.h:35,
                 from Marlin\src\HAL\HAL_LPC1768\../../core/../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\HAL_LPC1768\../../core/serial.h:24,
                 from Marlin\src\HAL\HAL_LPC1768\DebugMonitor.cpp:26:
Marlin\src\HAL\HAL_LPC1768\../../core/../inc/../core/boards.h:213: note: this is the location of the previous definition
 #define BOARD_BIGTREE_SKR_V1_3        2016  // BIGTREE SKR_V1.3 (Power outputs: Hotend0, Hotend1, Fan, Bed)
 
In file included from Marlin\src\HAL\HAL_LPC1768\../../inc/MarlinConfigPre.h:37,
[... 172 lines cut ...]
STM32F4                  IGNORED
STM32F7                  IGNORED
ARMED                    IGNORED
STM32F103VE_longer       IGNORED
mks_robin                IGNORED
mks_robin_lite           IGNORED
mks_robin_mini           IGNORED
mks_robin_nano           IGNORED
jgaurora_a5s_a1          IGNORED
STM32F407VE_black        IGNORED
BIGTREE_SKR_PRO          IGNORED
BIGTREE_BTT002           IGNORED
teensy31                 IGNORED
STM32F103CB_malyan       IGNORED
chitu_f103               IGNORED
teensy35                 IGNORED
esp32                    IGNORED
fysetc_f6_13             IGNORED
linux_native             IGNORED
SAMD51_grandcentral_m4   IGNORED
include_tree             IGNORED
========================================= 1 failed, 0 succeeded in 00:00:03.644 =========================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

maviles798, 2020-03-10:
i have the same problem, you were able to fix it?

Andy-ABTec, 2020-03-10:
`Marlin\src\HAL\HAL_LPC1768../../inc/../../Configuration.h:133: warning: "BOARD_BIGTREE_SKR_V1_3" redefined`

Don't know if its relevant but in the latest releases of Marlin the board definition is now:

#define MOTHERBOARD BOARD_BTT_SKR_V1_3

Check your boards.h file (Marlin > src > core > boards.h)

maviles798, 2020-03-10:
i try with the new and the old name and i have the same error

Andy-ABTec, 2020-03-10:
Sorry, in that case I have no idea...

dabossie, 2020-03-11:
Have you set LPC1768 in platformio.ini as your default environment?
[image]

maviles798, 2020-03-11:
I already fix the issue i miss type an aditional letter in the motherboard section, sorry guys, but thanks for you help. my firmware is ready and compiled. It is for an anycubic kossel plus with btt skr1.3 if you are interested just let me know.