bigtreetech/BIGTREETECH-SKR-V1.3 #189: SKR 1.4,  CR10S Bed Temperature Error

Ru83nG, 2020-01-07:
I get a "Printer Halted Error"  every time the bed temperature reaches 72°!!
Transitioned from the Creality board v2.1, where everything worked fine.  Now with this board I cannot figure it out.  I tried it with an external bed mosfet and with the on-board as well.  Same issue.
I checked the connection/polarity of the pins and all is as it was with the Creality 2.1.  
I am stumped that only at 72°, the bed temp will not go above that.
Checked the cabling to bed, then switched back to creality 2.1 and all works fine!!
Could anyone help?
Thanks in advance

Arganaz, 2020-01-09:
check your marlin code, you might have set a temperature limit or double check thermistor and bed connections for proper placement.
my bet is on the code.... switching between chamber and bed settings, maybe?
good luck

Ru83nG, 2020-01-12:
Thanks. done that also.  I think I remember marlin by heart now.I switched back to creality board 2.1 which works fine. I,ve put too many research hours into these boards.  Sent them, 1.3 and 1.4, back for refund.  With just a few days left in "return window." 
 I think I am ready to invest in something more solid and documented.

lijinlin19970813, 2020-01-14:
> Thanks. done that also. I think I remember marlin by heart now.I switched back to creality board 2.1 which works fine. I,ve put too many research hours into these boards. Sent them, 1.3 and 1.4, [...]

Can you send me your configuration.h / configuration_adv.h file to see?I can help you check.

closed 2020-02-14 by Ru83nG