bigtreetech/BIGTREETECH-SKR-V1.3 #191: Smoothieware

SankaMa, 2020-01-08:
Hello everyone, 

i've been trying to get smoothieware running on SKR v1.3 for a few days now, but i couldn't. With Marlin everything works quite well, this means there is no problem with the hardware. I stored the config.txt from Github in the SD card. I also get confirmation and feedback when I send G-code, but the motors are not rotating. Is it possible that the drivers are not working properly or do I need to make further changes in the config file? 

Thanks in advance.

sn0wrider, 2020-01-15:
Hi, it's not problem of drivers. developer smoothieware - wolfmanjm say "only original controllers are supported" 
if you want smoothieware, you need to find smoothieware like september-october 2018

vorpalk, 2020-03-03:
Then why is this board marketed as a Smoothieware compatible board? Does BTT provide a fork or not?

TWilmer, 2020-06-01:
Hi,
I would like to have a link to the source of firmware flashed to the controller initially, too.
Kind Regards
 Thorsten