bigtreetech/BIGTREETECH-SKR-V1.3 #192: TMC2209 and SKR 1.4 Turbo

delwinbest, 2020-01-10:
Hi Team,
I've spent a few hours trying to get the TMC2209 to register on the SKR 1.4 TURBO. I've tried 5 different drivers, and have followed your instructions. I've also tried your supplied firmware, and I have my own. Each time I get 'bad response'. 

Please help! I currently use these drivers in a MKS board without any trouble. 
My current config can be found here: https://github.com/delwinbest/Marlin_BIBO2_touch_dual/tree/btt_skr_v1.4_turbo

Attached is a photo of my jumper connection. I can't find anything else I can try. Could you perhaps provide a schematic for the Turbo board?

delwinbest, 2020-01-11:
Ignore. Found the problem: The board must be powered by main supply to test TMC drivers. Not USB. 5 hours lost :D

closed 2020-01-11 by delwinbest

wallen83, 2020-01-27:
so you can't check status in pronterface with only usb power? I spent 4 hours with it yesterday ...
also has problems with tmc2209 and sensorless homing .. it does not want to stop. can't find what's wrong. are you running sensorless homing?       (skr v1.4 board)

Fuglys, 2020-01-28:
waw thanks that can be the problem why M112 gave a kill

delwinbest, 2020-01-29:
@wallen83 that was my experience. I could not get it to work until I powered up the board with my 12/24V Supply. Also lost hours trying. 
I'm using sensor-less homing which works well.

thetechy2, 2020-02-04:
> also has problems with tmc2209 and sensorless homing .. it does not want to stop. can't find what's wrong. are you running sensorless homing? (skr v1.4 board)
1.4 turbo 2209 trouble with sensorless homing here please reply if youve found anything

wallen83, 2020-02-04:
[image]

yes i use sensor less on my skr1.4/TMC2009 UART.   I had removed // on the wrong sensor less homing in marling fw.  I had removed it on sensor_less_probing  🙈

you have all the pins left? (on TMC2009)

thetechy2, 2020-02-04:
> you have all the pins left? (on TMC2009)

yeah all the pins are there. when i try to home it, it just fails and halts the printer

drphil3d, 2020-04-20:
HEY GUYS! MAKE SURE TO CUT the pin on the Z stepper driver if your using tmc2209 RTFM!

drphil3d, 2020-04-20:
Also, I learned this one the hard way, if you enable sensorless homing on Z axis, the pins file changes the input from Z-STOP to PWR-DET port.