bigtreetech/BIGTREETECH-SKR-V1.3 #196: USB (mostly) stopped working

rally25rs, 2020-01-17:
My SKR 1.3 had been working fine for a couple days, with USB to both a Macbook and a Rasp Pi (octopi).

Suddenly it disconnected from OSX and ever since has been exhibiting weird behavior:

* If I plug it into any USB port on the Rasp Pi, it is not recognized (`dmesg` and `lsusb` show nothing conected). This had been working prior.
* If I plug it into my Macbook on the same port where it had been working, the board does not register.
* If I plug it into the _other_ USB port on my Macbook, it shows up as a drive (as it should) and I can see the SD card, however Printrun/Pronterface can't connect to it as a printer.
* The board does power up, on both 5v USB and 12v external PSU modes and the red light is on.
* The board will reflash itself if you put new firmware on the sd card (moves firmware.bin to firmware.cur as you would expect)

Testing with a meter, the data pins of the USB port do have continuity back to the pins of the main processor chip.
What really throws me for a loop is that it shows up as a drive on one USB port and not the others, but even when it shows up as a drive, it doesn't work as a printer.

I also tried connecting my Pi over the UART (to the Aux1 plug on the SKR) but that doesn't seem to work either. I can never get Octoprint to connect.

Using bugfix 2.0.x branch of Marlin (not the BTT fork) and the serial ports are set appropriately
```
#define SERIAL_PORT -1
#define SERIAL_PORT_2 0
#define BAUDRATE 115200
```
(and also didn't change between when this was working and stopped working)

I'm assuming something with the board or processor is toast, but... anyone have any thoughts on troubleshooting?

Marsman1970, 2020-01-18:
Well  i have  my serial port 2 set to 1  that worked better for me for some reason maybe worth trying.
Michael

Marsman1970, 2020-01-19:
Also if you have temperature sensor problems the usb stops working.

rally25rs, 2020-01-19:
I tried flipping to

#define SERIAL_PORT 2
#define SERIAL_PORT_2 -1

which is the default for the SKR minis
https://github.com/MarlinFirmware/Marlin/blob/2.0.x/config/examples/BigTreeTech/SKR%20Mini%20E3%201.2/Configuration.h#L106-L115

Interestingly my USB seems to be working again, even though I'm 99% sure that `-1` is the USB, and it was working fine as `SERIAL_PORT -1` before 🤷‍♂ 

So that gets me somewhere at least... I'd still like to figure out how to get my Pi connected over the serial pins on the AUX1 port.

I've also yet to find any actual docs on what the numbers for the serial port are... I've seen it listed that `0` is the AUX1 port, but as linked above, the example for the SKR minis are `2`, and you said you are using `1`... It'd be nice to know what the actual values represent 😕 

> Also if you have temperature sensor problems the usb stops working.

yeah I had noticed from a previous wiring mishap that if a limit switch is shorted then it halts the board too, so I have every connection disconnected while I was testing this. Actually the board is out of the printer just laying on the workbench on the static bag it came in with nothing but USB and the AUX1 pins for Gnd, TX, RX connected.

closed 2020-01-19 by rally25rs