bigtreetech/BIGTREETECH-SKR-V1.3 #197: Bigtreetech SKR V1.4 Turbo usb connection problem

Remohpant, 2020-01-20:
Hi, recently bought the skr 1.4 turbo and trying to upload marlin firmware 2.0.x. Issue is that pc doesnt recognise the usb connection. it fails after 1min of powering up the board. 
Does enaybody else encounter the same?

kaibs, 2020-01-20:
Do you have anything connected to the board?

I had a similar problem and in my case the reason for having the board crashing all the time was a damaged thermistor. It seems like if a sensor (endstops should cause this problem too) is shorted, marlin is prevented from booting and the board somehow kills it's USB connection.

Remohpant, 2020-01-20:
Thank you very much. It was thermistor. Once disconected them it worked.

closed 2020-01-22 by Remohpant