bigtreetech/BIGTREETECH-SKR-V1.3 #200: SKR 1.4 Documentation: Missing NPWR jumper description in the manual

stantond, 2020-01-21:
As titled. Please add what this jumper does to the manual.

csaintdenis, 2020-01-25:
yes, please do!

SusiBiker, 2020-01-29:
Another +1

Creality1968, 2020-02-07:
Another +2..

zLows, 2020-02-08:
C'mon, npwr - neopixel/bl/etc power supply selector (for dc-dc bridge owners). More clearely in pinout manual here https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT SKR V1.4/Hardware/BTT SKR V1.4PIN.pdf

stantond, 2020-02-08:
Still unclear on what it does. I have the DCDC module and bltouch. Which pins should this jumper be on?

zLows, 2020-02-08:
DCDC bridge converts 12/24 from VIN to 5 at VOT that means VOT+NPWR shoud be jumpered.

stantond, 2020-02-08:
Do you know how that differs from not using the bridge and leaving it on +5v?

zLows, 2020-02-08:
Oh, man... try to google something 'bout electronic schematics or so. :(
Your dcdc bridge useless 'till npwr jumpered with 5v.

stantond, 2020-02-09:
I understand that the DCDC bridge is a step-down converter configured to provide a continuous 3A at 5v, but I don't _know_ what the point of it is on this board. I've seen others on YouTube using a BL Touch, with the DCDC module, but with this jumper unchanged (not set as you've suggested). I _assume_ that the board itself can handle this one device easily, but it would not handle supplying consistent current to multiple 5v devices, whereas the DCDC bridge does enable this, but I don't want to _assume_, I want to _know_, and I expected the function, limitations, and jumper settings to be described in the documentation (or at least the product listing for the module on Ali!), which is not unreasonable.

I bought it because the difference between the cost of the board standalone vs the board with several optional modules was negligible, so like anyone with a bit of foresight when buying electronics from China, I ordered them together, so I wouldn't discover I needed a module later and have to wait another month for it to arrive before I can resume printing.

Clyde78, 2020-03-13:
From my understanding, the DCDC bridge is a buck converter that provides 5V for the TFT, BLTouch and Neopixel headers. The SKR 1.4 got its own onboard 5V converter, but it's shared with whatever else on the board needs 5V, so if you plan on running a lot of neopixels or other 'heavy' 5V loads, getting the DCDC bridge is adviced. With that module plugged in, you can draw up to 3A on the 5V rail. I'd recommend staying beneath that top load however. 2.0-2.5A should be fine. The manual advices to run less than 30 neopixel LEDs, which is plenty.  
The NPWR jumper selects the 5V source for the TFT, BLTouch and Neopixels. With the jumped in its default position (bottom and middle pin) the onboard 5V regulator is used. If you got the DCDC bridge, move the jumper to the top and middle pin to actually USE the bridge too.

sl1pkn07, 2020-04-11:
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V1.2/The%20Notement%20of%20BTT%20SKR%20MINI%20E3%20V1.2.pdf

avidNewb, 2020-07-16:
I appreciate those of you who answered this question reasonably and without criticism.  Your answers were just what I needed.  As someone who creates documentation daily, it is fair to say the documentation for these boards and parts is lacking.  Any abbreviations, jumpers, and pins should be clearly defined.

Shifuman, 2020-12-12:
Hello, I have just tested to **remove** the jumper : my BLTouch AND my TFT are still powered, so I don't understant what is the purpose of this jumper J9 !

If this is really a jumper to select the power source, why it remain powered after removing the jumper ?

wrych, 2021-02-05:
For those like: https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.4/Hardware/SKR-V1.4-Turbo-pinout.jpg

thebeline, 2022-05-01:
Not to revive a dead topic, but another consideration:

When using a Raspberry Pi, it is advisable to only connect GND/Tx/Rx and leave 5v unbonded (as it can cause issues with the onboard 5v regulator (or separate regulator).  That said, you could remove the `NPWR` jumper entirely, and bond the 5v from the Pi to the `NPWR` pin on the `TFT` connector.  This would cause `NPWR` to be provided 5v from whatever the Pi was receiving 5v from (it is a direct passthrough).

For those of us without the external buck converter, but a high-power supply for the Pi, this is a very useful trick.

yasaslive, 2022-08-29:
Can we close this issue?

Zfrenchy, 2024-02-17:
[image]