bigtreetech/BIGTREETECH-SKR-V1.3 #202: How to Turn on Fan automatically when temperature goes to 50. Which pins i should define. I have SKR 1.4 Turbo

Talha909, 2020-01-23:
How to Turn on Fan automatically when temperature goes to 50. Which pins i should define. I have SKR 1.4 Turbo. I also saw the pin file.

ellensp, 2020-01-24:
If you saw the pin file then you also saw that FAN1 is directly connected to power and is not software controllable. So you cannot use the FAN1 plug. You have to use something else. If you only have 1 hotend then the firmware sets FAN1_PIN to use the HE1 connector.  This is software controllable but to use it with auto fan you need to stop the pin being used by FAN1_PIN. Add "#define FAN1_PIN -1" to your configuration.h and setup autofan by setting E0_AUTO_FAN_PIN pin to P2_04 in Configuration_adv.h  If you have more than 1 hotend... you need to add external mosfets

Talha909, 2020-01-25:
but there is one cnc fan

Evanlec, 2020-02-12:
There are 3 fan headers on the SKR 1.4 -- but only one is PWM? (i.e. the CNC fan header?).

I'm not using a part cooling fan, but would like hot-end fan to only come on over 50C. With SKR v1.3 I used the HE1 header for this, as mentioned, but I thought maybe I wouldn't need to use that trick with v1.4?

Sounds like that's not the case with SKR v1.4...which is disappointing...why did they add more always-on fan headers? Not very useful to me (I don't want fans spinning all the time when the printer is sitting idle, super annoying)

ellensp, 2020-02-12:
I completely didn't seen fan2. 
But yes it is also just power and gnd, no software control on that one either..  
There are ways around this. Add external mosfets etc. But the simplest is just attach one of these to your hotend and put in series with the fan + line. https://www.aliexpress.com/item/32658237257.html
Is a thermal switch off until > 50c then turn on. Often used as fan 'controllers' in cheap power supplies.

drphil3d, 2020-04-19:
> There are 3 fan headers on the SKR 1.4 -- but only one is PWM? (i.e. the CNC fan header?).
>
> I'm not using a part cooling fan, but would like hot-end fan to only come on over 50C. With SKR v1.3 I [...]

The most disappointing part about the SKR 1.4 Turbo is out of the 4 Fan headers ONLY ONE HAS PWM control. WTF? Why? I thought Turbo would be better but no, I still have to listen to the Turbo fans anytime the printer is turned on. I'm doing dual extrusion so I can't use HE1 for fan control. Time to make my own fan control board... :(

kkosa1, 2020-04-19:
Have you set
#define E0_AUTO_FAN_PIN
to your appropriate fan pin as shown in the boards_pinns file?  I am using an SKR Pro and it works great.?  this in in configuration_adv.h around line 422.  then you can set your desired "on" temp
#define EXTRUDER_AUTO_FAN_TEMPERATURE
at line 430

drphil3d, 2020-04-20:
yes, that's already enabled. The SKR pro is a totally different board. The problem is the pins are connected directly to power and GND. just like the schematic shows. What I'm not happy about is the sales information for the SKR 1.4 Turbo leaves the reader with the impression that there is more than 1 Fan mosfet. It's misleading.
[image]

Evanlec, 2020-04-21:
> > There are 3 fan headers on the SKR 1.4 -- but only one is PWM? (i.e. the CNC fan header?).
> > I'm not using a part cooling fan, but would like hot-end fan to only come on over 50C. With SKR v1.3 I used the HE1 header for this, as mentioned, but I thought maybe I wouldn't need to use that trick with v1.4?
> > Sounds like that's not the case with SKR v1.4...which is disappointing...why did they add more always-on fan headers? Not very useful to me (I don't want fans spinning all the time when the printer is sitting idle, super annoying)
> 
> The most disappointing part about the SKR 1.4 Turbo is out of the 4 Fan headers ONLY ONE HAS PWM control. WTF? Why? I thought Turbo would be better but no, I still have to listen to the Turbo fans anytime the printer is turned on. I'm doing dual extrusion so I can't use HE1 for fan control. Time to make my own fan control board... :( We have all these other fancy features that I will probably never use but we can't get more than 1 mosfet for fans? come on guys it's 2020 we're better than this.

I agree -- I was totally dismayed to find out that this board has extra fan headers, but only **one** of which is PWM...which is the same as V1.3.

Non-PWM fan headers are not very useful to me, unless I feel like listening to my fans all the time while the printer isn't doing anything.

drphil3d, 2020-04-21:
exactly, pisses me off. I can tell by looking at the schematics the way
they designed it would require adding another buffer just to control the
fan mosfets so they probably just skipped it.

Something like this would offer a solution. This RRD fan extender has a max
voltage of 20v
http://www.geeetech.com/wiki/index.php/Reprap_Ramps1.4_RRD_Fan_Extender

EagleeyeKai, 2020-04-23:
Shouldn’t it be possible to use the HE1 / P2_04 as a switchable hotend Fan output? 
At least that’s what I tried today and configured the autofan option with P2_04 on Marlin 2.0.4.4 However, it did not work. Need to investigate tomorrow but would be happy to get further input as well.

drphil3d, 2020-04-24:
sure, but that defeats having dual extrusion which I do

ngathant, 2020-05-08:
> 
> 
> Shouldn’t it be possible to use the HE1 / P2_04 as a switchable hotend Fan output?
> At least that’s what I tried today and configured the autofan option with P2_04 on Marlin 2.0.4.4 However, it did not work. Need to investigate tomorrow but would be happy to get further input as well.

Did you get yours to work? I cut my fan headers and powered them on HE1 and set #define E0_AUTO_FAN_PIN P2_04 and #define EXTRUDER_AUTO_FAN_TEMPERATURE 50 but noticed the fan isn't turning on even at 230'C.

Epic14180, 2020-05-21:
Same problem set #define E0_AUTO_FAN_PIN P2_04 and #define EXTRUDER_AUTO_FAN_TEMPERATURE 50 but noticed the fan isn't turning on

acinsa, 2020-06-11:
Hi, I have exactly the same problem. I can run the hot end fan from the LCD using the FAN Speed 2 setting under Temperatures. This indeed is turning on the Mosfet for HE1. But to get it to turn on automatically does not work.

spiffy781, 2020-07-08:
Ok still no update, but I do see configuration.adv.h   #if defined(BTTSKRE3MINIV2_0) 
#define USE_CONTROLLER_FAN  

[image]

[Archive.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/4888162/Archive.zip)

I am ender 3 / skr 1.4 T,2208, TFT, octopi,  

Just trying to keep this alive there is got to be too simple of an answer M106 , etc, any help would be greatly appreciated.

ngathant, 2020-07-08:
I got it to work using HE1 and #define E0_AUTO_FAN_PIN P2_04. It's been a couple of months. Traced my issue to incorrect wiring. My Creality CR10S has a radial connector for all wires going to the hot end. Turned out black & red cables coming out from the board was yellow yellow on the other end going to the HE. You gotta love Creality's attention to detail. Plug in a fan with short wires so you know there's no issue there if you have radial connectors. It's likely the same issue in an Ender 3. If you have something else, it's very likely a wiring issue. Plugging your fan to HE1 and setting that P2_04 should work for you unless you have dual extruders in which case skr 1.4 may not be the board for you.

spiffy781, 2020-07-08:
Ok, so quick update, I am running (Chocolate Marlin) if you're here with fan issues, you are maybe like me and trying to program you BigTreeTech SKR 1.4 (turbo) LPC1768/9, environments.  Flashing with cheetah 5.0 ( thank you KAY3D) anyhow, here is the code.   ( it helped me )   

https://www.mycomputerwizards.com/AlexB2/cheetah/-/commit/f7e35f211d38ad3ef11f027543fa9bda3473897c

[image]

spoiler alert it's at the bottom of the code 😂 
`// Section - Custom Codes`

`E0_AUTO_FAN_PIN FAN1_PIN`

Good luck,  thanks to KAY3d, and tripod garage, YT.

Fireman68, 2020-07-26:
Hello. Can I connect two coolers in parallel in one controlled socket to cool a part? Skr 1.4 turbo board.

Eddiiie, 2020-08-19:
I have 2 extruders and 4 fans.   On my old RAMPS board I had this working.
It looks like I can re-use some other pins that I won't use like the UPS or Filament sensor.
My question is:  Are these pins PWM capable?
Are all the pins on the 1.4 Turbo  PWM capable?

Alexkvw, 2020-10-29:
Anyone solves this? Not all pins are PWM capable. from what i can tell (bit of a n00b) the chip has 1 PWM modulator with 6 channels. Channel 1,4,5,6 are in use for H1, H1, Bed and Fan which leaves PWM channel 2 and 3 unused. These should be available on pins P1.20/P2.1 and  P1.21/P2.2 respectively. The P2.1 and P2.2 pins are in use for the X axis stepper leaving P1.20 and P1.21 which are both broken out trough EXP1. 

Can anyone confirm this reasoning?

Eddiiie, 2020-10-30:
I have 2 extruders.  1 fan on each extruder.     

My Marlin configuration_adv.h   for auto temp says

```
#define E0_AUTO_FAN_PIN P0_00
#define E1_AUTO_FAN_PIN P0_01

#define EXTRUDER_AUTO_FAN_TEMPERATURE 50

```

I use the pwm signal on those pins  (full on  or full off) to an external MOSFET board with 4 FETs on it.
This pwm signal is not strong enough to drive a fan by itself, of course.

I also have 2 part cooling fans.   Still struggling with that.   I want to use the fan MUX board.  Just need to tinker with it some more.   Not really happy with 2 extruders.  GrrrrrRRRRRrRrRrRRr

Hope this helps..  I use  pins p0_00  and p0_01 for auto temp pwm signal.    You dont really need PWM on an extruder fan.  YOu need it on the part cooling fan.

Alexkvw, 2020-10-30:
Ahh of course! not using PWM for the extruder fan makes life a lot easier.

mpselectrolab, 2021-02-01:
This is very frustrating, but the limitation is not room for FET, but the LPC1769 support enough PWMs. They feed PWM to the fans for setting the speed and best energy control of the brushless DC fans. Running fans wide open at their rated voltage is usually far to much cooling for the hotends, so driving a FET straight ON/OFF was not considered, I would guess. I would not do that either. They hit a design impasse on this with the LPC1769 and faced choosing what options to give. However, giving the ability to support dual extruders/hotends and not provide this is a bit interesting. I actually like this board and just hit this on updating my FlashForge to this controller. I hit two things, need separate thermocouple amp and need to add on some fan control for the hot-end fans. Working on a board that does both of these in one swoop.

carlosjln, 2021-11-06:
I hope this sheds some light.

I'm new to Marlin and SKR boards, for the past 7 years I've been "stuck" with smoothieware and azteeg boards and this feature was achieved straight forward on the config.

After watching some videos got around `E0_AUTO_FAN_PIN` but setting it to use `P2_03` would give me an error while building because it was being used by `FAN_PIN`. I disabled it as you can see on the attached screenshot and the build proceeded. 

I think tomorrow I'll be able to test it with the hotend connected.

[image]

mccunnj, 2023-12-20:
I got dual HE fans set to auto on my SKR 1.4.[I made a video explanation](https://youtu.be/NSjnlr60vZE)