bigtreetech/BIGTREETECH-SKR-V1.3 #203: Troubleshooting MOSFET driven hotend fan

eschmidt12, 2020-01-24:
I'm in the process of upgrading my printer to a dual extruder setup with the TH3D Tough dual hotend, Petsfang mount, and the SKR V1.4 main board running 2209 drivers.

The issue I'm having is that, because I'm now using both HE0 and HE1 outputs for extruders, I don't have a g-code controlled hotend cooling fan. I could run the fan easily enough all the time, but I really want to be able to use the auto fan capability, i.e. only on when the hotend is over 50C.

Here's the configuration I'm trying to troubleshoot now:

* Marlin 2.0.x bugfix
* IRF520 MOSFET module driving the fan:
   * VIN and GND from the PSU
   * V+ and V- connected to the fan
   * SIG and GND connected to the E0DET pins 1.26 and GND. E0DET is typically configured as the filament runout sensor, and is right next to the x, y, and z endstop pins.
* In Configuration\_adv.h:
   * \#define E0\_AUTO\_FAN\_PIN P1\_26
* In Pins\_BTT\_SKR\_V1\_4.h:
   * //#define FIL\_RUNOUT\_PIN     P1\_26
   * \#define FAN1\_PIN          P1\_26
* In Configuration.h:
   * \#define ENDSTOPPULLUPS

Here's the problem:

What I'm finding is that the MOS module is behaving correctly if I drive its SIG pin with a known 3.3/5V source (i.e. the +5V pin on E0DET). The fan runs and everything, no worries. But I can't get the SIG pin on the SKR V1.4 to output a voltage to drive the MOS module.

* If I measure the voltage between +5 and GND at E0DET, I get +5V, as expected
* If I measure the voltage between SIG and GND at E0DET, I get \~20mV when I expect it to be in the "off" state
* If I measure the voltage between SIG and GND at E0DET, I get \~240mV when I expect it to be in the "on" state. I can observe the voltage change occur exactly when the AUTO\_FAN\_PIN configuration should trigger it, when the hotend temperature crosses the 50C threshold.
* If I measure the voltage between the SIG and GND pins for E1DET and PWRDET, I get somewhat random readings. If I measure between SIG and GND for X, Y, and Z-stop pins, I get a solid 3.3V.

So the E0DET pin seems to be changing state as expected, but the "on" voltage of \~240mV is a small fraction of what I expect it to be, therefore it can't trigger the MOSFET.

What am I missing? Do I have a defective board? An incorrect configuration of the firmware? Are those the wrong pins to use? I'd love some help figuring this one out.

OleUrgast, 2020-01-29:
If I remember correctly, all endstop connectors have hardware resistors and capacitators on SKR boards.  You actually send your signal through a voltage devider if using them as output. While this is wanted on input (as I/O of CPU is 3.3 V but input should be 5V (or even 5.5V) tolerant) it also reduce 3.3V to a value to low to switch a mosfet. This a disadvantage in using a 32bit CPU with 3.3V instead 5V 8 bit Arduino...
Depending on your display, some pins on EXP1/2 should be free to use (i.e. if you do not use an 2004 LCD). These have no resistors attached, as they are output anyway (even displays with 5 V accept 3.3V as high) so they should work fine (have not testet myself).

eschmidt12, 2020-01-29:
> If I remember correctly, all endstop connectors have hardware resistors and capacitators on SKR boards. You actually send your signal through a voltage devider if using them as output. While this is [...]

Thanks for the feedback. You might be right, but in that case why would I see different behavior between the x, y, and z min connector pins, and the E0DET, E1DET, and PWRDET pins? I get 3.3V to the 3 latter, measured at the pins.

I’m running the stock Ender 5 LCD...any idea if that has available EXP1/2 pins?

OleUrgast, 2020-01-29:
It might be Marlin switching internal pullup/pulldown resistors in addition to the hardware ones...

in https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.4/Firmware/Marlin-2.0.x-SKR-V1.4-Turbo/Marlin/src/pins/lpc1769/pins_BTT_SKR_V1_4_Turbo.h you find pin definitions used for different displays; beginning in line 178.
Ender 5 uses CR10 display. So if you use only one ribbon cable to exp3 of display, connect it to exp1 on board and activate cr10 display in marlin, the standard definition of exp1 / exp2 is changed by:
```
  #if ENABLED(CR10_STOCKDISPLAY)
    #define LCD_PINS_RS    P1_22
    #define BTN_EN1        P1_18
    #define BTN_EN2        P1_20
    #define LCD_PINS_ENABLE P1_23
    #define LCD_PINS_D4    P1_21
```

Normal setup at EXP1/2 is:
```
 *              _____                                             _____
 *          NC | · · | GND                                    5V | · · | GND
 *       RESET | · · | 1.31(SD_DETECT)             (LCD_D7) 1.23 | · · | 1.22 (LCD_D6)
 *  (MOSI)0.18 | · · | 3.25(BTN_EN2)               (LCD_D5) 1.21 | · · | 1.20 (LCD_D4)
 * (SD_SS)0.16 | · · | 3.26(BTN_EN1)               (LCD_RS) 1.19 | · · | 1.18 (LCD_EN)
 *   (SCK)0.15 | · · | 0.17(MISO)                 (BTN_ENC) 0.28 | · · | 1.30 (BEEPER)
 *              -----                                             -----
 *              EXP2                                              EXP1
```
So if you connect the display by only one ribbon cable and define CR10 stock display in Marlin, EXP2 is completly free. Not shure if 0.15-0.17 is not used internaly otherwise (does not seem so), but 3.25+3.26 should be free in my opinion.

eschmidt12, 2020-02-02:
> It might be Marlin switching internal pullup/pulldown resistors in addition to the hardware ones...
>
> in [...]

Implemented using 3.25 and 3.26, everything working perfectly. Thanks!

closed 2020-02-02 by eschmidt12