bigtreetech/BIGTREETECH-SKR-V1.3 #206: SKR 1.4 Turbo USB Detection

Baconadors, 2020-01-25:
Hi all, 

Just getting my SKR 1.4 Turbo wire up and I seem to be having a problem connecting it via USB.
I have changed the jumper over to USB power. I plug in the provided USB cable to the board and then to my computer.

However it will not show up in explorer.

I check the device manager and I can see a "USB Serial Device (COM4) flashing every few seconds.
I know that this is the SKR board as this behavior stops when I unplug it.

What is the issue? See the attached GIF.
[image]

bartofborg, 2020-01-25:
You need the driver for it, https://github.com/HobbyComponents/CH340-Drivers

Baconadors, 2020-01-25:
Unfortunately, that driver won't install.

https://i.imgur.com/xpB9M7L.gif

bartofborg, 2020-01-25:
if you open the instruction pdf, ther is a link in it to a driver from bigtree there. it is a dropbox address, you'll have to type it in manually but, it worked for me

Baconadors, 2020-01-25:
I disabled signed driver checking and it allowed me to install both the CH340 driver and the lpc176x_usb_driver included in github.

Still have the same issue with it showing. Same behavior as in the original gif.

Baconadors, 2020-01-25:
If I uninstall the driver right before it disappears, when the device reappears it will show "Marlin USB Device"
It this disappears and the cycle in the original gif continues. At this point I have indeed installed the drivers.

[image]

Baconadors, 2020-01-25:
When I right click the device and choose update, I was able to switch it over to Marlin USB Serial driver.
[image]

However, original behavior continues.

bartofborg, 2020-01-25:
Good! You ready to go then.Jeff Zackodnik

Baconadors, 2020-01-25:
No, I'm not ready to go.

It still doesn't work.

Baconadors, 2020-01-25:
[image]

bartofborg, 2020-01-25:
Have you connected the USB to your computer and opened pronterface or whatever terminal program your using and tried connecting to the board on the com port the marlin device is?

https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/206#issuecomment-578436223

Baconadors, 2020-01-26:
Wiring issue.

Couldn't directly see what was causing a problem, but I redid all of the wiring and made sure to put electrical tape on anything that could short the board.

closed 2020-01-26 by Baconadors

PapiOrso, 2020-02-10:
I solved the problem by reloading the original firmware provided in the micro SD. I used a usb card reader to rename the original file from .CUR to .BIN then I inserted the SD in the board.

saintofinternet, 2021-01-19:
> I solved the problem by reloading the original firmware provided in the micro SD. I used a usb card reader to rename the original file from .CUR to .BIN then I inserted the SD in the board.

Does that orginal firmware work for CoreXY?

lakeland3516, 2021-03-02:
the reason the inf file wont' work for people using windows 10 or other windows i found that the 64 bit machines won't recognize the file but if you have a old 32 bit computer kicking around it'll load properly and recognize it so basically saying some one needs to write one for a 64 bit system