bigtreetech/BIGTREETECH-SKR-V1.3 #207: SKR v1.4 Turbo + BLTouch marlin configuration

flyhigh-hifi, 2020-01-26:
Initial marlin configuration has **Z_MIN_PROBE_PIN** and I don't see in manual what should it be. Currently my BlTouch doesn't work because of that I guess.

Evanlec, 2020-01-27:
I'm having the same issue. I tried just commenting out this line:
`// #define Z_MIN_PROBE_PIN 32 // Pin 32 is the RAMPS default`
and also leaving this line commented out:
`// #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`

The probe can do self-test etc, but when I try to home the Z axis, the BLTOUCH probe does not deploy.
When the BLTOUCH is sitting idle, an M119 command shows that Z_Probe is in state 'TRIGGERED' , when they should not be.

`Send: M119
Recv: Reporting endstop status
Recv: x_min: open
Recv: y_min: open
Recv: z_min: open
Recv: z_probe: TRIGGERED
Recv: ok`

oldman4U, 2020-01-28:
Hi.

I am new to 3D printing and Marlin but have been able to get the BLTouch working on my SKR Pro v1.1, so maybe i can help a bit.

A assume you use and end stop device for X, Y and Z and the BLTouch in addition to them.

Using the dedicated Probe connector on the mainboard, the "define" values in Marlin to use the end stop pin have to be commended out, so there has to be a // in front of the two lines shown above starting with #define Z_MIN_Probe_... and "define BLTouch" has to be used. This is also shown in the pins file, like you can see in the screenshot.
[image]

Hope this helps

flyhigh-hifi, 2020-01-28:
> Hi.
> 
> I am new to 3D printing and Marlin but have been able to get the BLTouch working on my SKR Pro v1.1, so maybe i can help a bit.
> 
> I assume you use an end stop device for X, Y and Z and the BLTouch in addition to them.
> 
> Using the dedicated Probe connector on the mainboard, the "define" values in Marlin to use the end stop pin have to be commended out, so there has to be a // in front of the two lines shown above starting with #define Z_MIN_Probe_... and "define BLTouch" has to be used. This is also shown in the pins file, like you can see in the screenshot.
> [image]
> 
> Hope this helps

Thanks for your addition. Bltouch works on my SKR 1.4 when I'm using Z_MIN_ENDSTOP  as bltouch probe pin, but there's new port on SKR 1.4 Turbo designed exactly for Bltouch, would be nice to use it as it was designed.

S13Tyler, 2020-02-02:
IDK if this helps but I pulled an all-nighter getting my SKR V1.4 (non-turbo) all configured for my printer and first time installing BLTouch. I managed to get the BLTouch pin headers on the board to work by making a few changes. I started with a fresh 2.0.x branch since I couldn't get the BTT version to compile.

Comment out the line:
`//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`

and define:
`#define Z_MIN_PROBE_PIN P0_10`

Open Conditionals_LCD.h located in Marlin\src\inc\ and change these lines:

to:
`#define HAS_CUSTOM_PROBE_PIN  true // DISABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)
#define HOMING_Z_WITH_PROBE   (Z_HOME_DIR < 0) // (Z_HOME_DIR < 0 &&!HAS_CUSTOM_PROBE_PIN)`

S13Tyler, 2020-02-02:
IDK if this helps but I pulled an all-nighter getting my SKR V1.4 (non-turbo) all configured for my printer and first time installing BLTouch. I managed to get the BLTouch pin headers on the board to work by making a few changes. I started with a fresh 2.0.x branch since I couldn't get the BTT version to compile.

Comment out the line in **Configuration.H**:
`//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`

and define:
`#define Z_MIN_PROBE_PIN P0_10`

Open **Conditionals_LCD.h** located in ..Marlin\src\inc\.. and change these lines:
`#define HAS_CUSTOM_PROBE_PIN  DISABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)`
`#define HOMING_Z_WITH_PROBE   (Z_HOME_DIR < 0 &&!HAS_CUSTOM_PROBE_PIN)`

to:
`#define HAS_CUSTOM_PROBE_PIN  true`
`#define HOMING_Z_WITH_PROBE   (Z_HOME_DIR < 0)`

Basically Marlin doesn't let the printer home the Z-axis with the probe when an end-stop is connected. When I get time, I'll mess around some more and try to add some lines so that the end-stop can be used as a fail-safe in case of probe failure (which is probably the original code's intent). 

My BLTouch is fully functional and tested with this method. Just keep an eye on it because I'm not sure if the end-stop will do anything if the probe doesn't home properly.

Evanlec, 2020-02-03:
> IDK if this helps but I pulled an all-nighter getting my SKR V1.4 (non-turbo) all configured for my printer and first time installing BLTouch. I managed to get the BLTouch pin headers on the board [...]

Thanks for this work. I shall try it out as soon as this current print is finished.

One question just to clarify:

This is when using the designated BLTOUCH probe pins, i.e. these:
[image]

and **not** plugging the probe into the Z-min stop header (as was required with SKR v1.3) _correct_?

Also, should it not be possible to have both the BLTOUCH and the stock Ender 3 Z end-stop (it being plugged into Z-Min header) working as well, to provide as you said, a failsafe?

hopeinformer, 2020-02-03:
That is very interesting. I will give that a try.

I am having a similar issue. However, I did finally fix the BLTouch not
homing Z by using the Z-endstop pins for it instead of the designated probe
pins. It worked as it should for a few times then went right back to
crashing into the bed. I didn't change the firmware at all between it
working and it not working. So this leads me to believe the actual
motherboard SKR V1.4 has an issue. I can't figure it out. I'm thinking of
just ditching the whole BTT line and going with a Duet instead.
On 2/2/20 11:53 PM, Evan LeCompte wrote:

IDK if this helps but I pulled an all-nighter getting my SKR V1.4
(non-turbo) all configured for my printer and first time installing
BLTouch. I managed to get the BLTouch pin headers on the board to work by
making a few changes. I started with a fresh 2.0.x branch since I couldn't
get the BTT version to compile.

Comment out the line in *Configuration.H*:
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

and define:
#define Z_MIN_PROBE_PIN P0_10

Open *Conditionals_LCD.h* located in ..Marlin\src\inc.. and change these
lines:
#define HAS_CUSTOM_PROBE_PIN DISABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)
#define HOMING_Z_WITH_PROBE (Z_HOME_DIR < 0 &&!HAS_CUSTOM_PROBE_PIN)

to:
#define HAS_CUSTOM_PROBE_PIN true
#define HOMING_Z_WITH_PROBE (Z_HOME_DIR < 0)

Basically Marlin doesn't let the printer home the Z-axis with the probe
when an end-stop is connected. When I get time, I'll mess around some more
and try to add some lines so that the end-stop can be used as a fail-safe
in case of probe failure (which is probably the original code's intent).

My BLTouch is fully functional and tested with this method. Just keep an
eye on it because I'm not sure if the end-stop will do anything if the
probe doesn't home properly.

Thanks for this work. I shall try it out as soon as this current print is
finished.

One question just to clarify:

This is when using the designated BLTOUCH probe pins, i.e. these:
[image: image]
<https://user-images.githubusercontent.com/10586/73634800-02594200-4630-11ea-815a-910f9334f5e1.png>

and *not* plugging the probe into the Z-min stop header (as was required
with SKR v1.3) *correct*?

Also, should it not be possible to have both the BLTOUCH and the stock
Ender 3 Z end-stop (it being plugged into Z-Min header) working as well, to
provide as you said, a failsafe?

Evanlec, 2020-02-04:
Thank you very much @S13Tyler , that did the trick!

S13Tyler, 2020-02-04:
@Evanlec No problem man! As a warning though this is just a band-aid fix. As soon as I have time to sit down and write some new lines I'll post it for you guys or maybe someone else will dive in beforehand!

thelittlemike, 2020-02-08:
> 
> 
> @Evanlec No problem man! As a warning though this is just a band-aid fix. As soon as I have time to sit down and write some new lines I'll post it for you guys or maybe someone else will dive in beforehand!

Have you got this working using the dedicated pins yet? I've been asking around, but no one seems to have found a solution.

stantond, 2020-02-11:
Good work @hopeinformer ! I've just run into this myself and you've saved me a lot of pain. For anyone else reading through this:

> Comment out the line in *Configuration.H*:
> //#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

Let's assume this is set correctly by the user. This should be commented out on any board, including the SKR 1.4/Turbo, where there's a dedicated pin/port for the probe.

> and define:
> #define Z_MIN_PROBE_PIN P0_10

This isn't required - it's already set correctly in the appropriate pins file.

> Open *Conditionals_LCD.h* located in ..Marlin\src\inc.. and change these
> lines:
> #define HAS_CUSTOM_PROBE_PIN DISABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)
> #define HOMING_Z_WITH_PROBE (Z_HOME_DIR < 0 [...]

No need to modify the first line. If configured correctly by commenting out `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` because we have a dedicated probe pin, the `DISABLED` function here returns true, so `HAS_CUSTOM_PROBE_PIN` is `true`. That's all fine and correct. We agree that we have a custom probe pin, and the pins file also agrees.

Now, that last line... `Z_HOME_DIR < 0` is fine, this ensures you're probing downwards for your Z axis, which we are, and by default this is set to `-1`. So for most users, this returns `true` and we can ignore it here.

However, `!HAS_CUSTOM_PROBE_PIN`... This will be `true` if we *do not* have a custom probe pin (the `!` causes the inversion). As we *do* have a custom probe pin, this is `false`, so `HOMING_Z_WITH_PROBE` is also set to false.

In essence, the Marlin developers have said, "If you have a custom probe pin, you cannot home with your Z probe." I can't believe this is desired behavior, more of an oversight, so I'll raise this as a bug.

Tl;dr: in `Marlin/src/inc/Conditionals_LCD.h`, replace:

`#define HOMING_Z_WITH_PROBE (Z_HOME_DIR < 0 &&!HAS_CUSTOM_PROBE_PIN)`

with

`#define HOMING_Z_WITH_PROBE (Z_HOME_DIR < 0)// &&!HAS_CUSTOM_PROBE_PIN)`

emaayan, 2020-02-15:
i've managed to reproduced as well , correcting 
  #define HOMING_Z_WITH_PROBE   (Z_HOME_DIR < 0)// && !HAS_CUSTOM_PROBE_PIN)
and commenting //#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

this is a rather dangerous band-aid, if you just disable Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN the probe will come crushing down the bed, so this fix could be forgotten during upgrades between firmwares!

PapiOrso, 2020-03-16:
**@S13Tyler** - The change you suggested in the Conditionals_LCD.h file works great until version 2.0.4.4. of Marlin.
Unfortunately in the new release2.0.5 the Conditionals_LCD.h file is very different.
Can you still help us? Thank you very much!

thelittlemike, 2020-03-16:
So I got the BLTouch (3.1 version) to work just fine making no edits whatsoever to vanilla Marlin other than enabling BL Touch and 5V and a single change to the pins file:

#else
  #define Z_STOP_PIN       P0_10   // Z-STOP
#endif

This uses the dedicated pins on the board, with the hotend fan plugged into HE1 and the cooling fan plugged into FAN0

nakedandjameson, 2020-03-16:
@thelittlemike Did you get this working with the v1.4 turbo? Been having a lot of difficulty with trying to get my BLTouch set up with no luck. I tried the fix above (commenting out second half of HOMING_Z_WITH_PROBE) and that didn't work for me. I'll try modifying the pins.h file like you did and see what happens.

Also, if it is the v1.4 turbo, shouldn't BLTouch be in 3.3v mode since the v1.4 turbo has 3.3v logic? Why set it to 5v mode?

thelittlemike, 2020-03-16:
> @thelittlemike Did you get this working with the v1.4 turbo? Been having a lot of difficulty with trying to get my BLTouch set up with no luck. I tried the fix above (commenting out second half of [...]

This was on the SKR 1.4 Turbo (LPC1769).

Are you sure that it is only 3.3V logic? Some of the pinouts denote 5v. Unfortunately the schematic does not say what the probe puts out and I haven't actually hooked up a multimeter to it to confirm one way or the other. I can say that the probe is working just fine after my changes, though. I'm even doing a 7x7 (49 point) grid on my ABL; which takes forever, honestly. Not sure if I recommend that.

nakedandjameson, 2020-03-16:
@thelittlemike I'll try changing it to 5v and see what happens. I just tried your fix with 3.3v. When I auto home, X and Y home correctly and Z starts moving down, but the probe doesn't deploy.

thelittlemike, 2020-03-16:
> 
> 
> @thelittlemike I'll try changing it to 5v and see what happens. I just tried your fix with 3.3v. When I auto home, X and Y home correctly and Z starts moving down, but the probe doesn't deploy.

I tried with both force 5v on and off and it worked both ways for me, honestly. The only time it *didn't* work was before I set that Z_STOP pin. Other than that, it's been working great. One thing you can try is to use Cheetah 5.0, which is a build of Marlin 2.0.3 and change the pin file.

nakedandjameson, 2020-03-16:
@thelittlemike Tried with 5v and your fix. My pin still doesn't deploy for whatever reason during Z homing, even though it works fine with deploy/stow and self-test.

thelittlemike, 2020-03-16:
> 
> 
> @thelittlemike Tried with 5v and your fix. My pin still doesn't deploy for whatever reason during Z homing, even though it works fine with deploy/stow and self-test.

Did you try Cheetah?

cronos45, 2020-03-16:
> I tried with both force 5v on and off and it worked both ways for me, honestly. The only time it _didn't_ work was before I set that Z_STOP pin. Other than that, it's been working great. One thing [...]

Cheetah does nothing special.

thelittlemike, 2020-03-16:
> Cheetah does nothing special.
That's a bit of a stretch.

If my configuration works and nakedandjameson tried making the same changes and it didn't work for them, using Cheetah could help in that it would make sure they have everything else configured correctly. I mentioned Cheetah because it works. I'm running it right now while I work on upgrading to 2.0.4.4.

xkuyax, 2020-03-20:
Whats the current status?
I got my new v1.4 Turbo and i got everything to work except my Trianglelab BLTouch. Im currently on the newest Marlin 2.0.5

If i disable Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN my printer doesnt home Z, if i enable it i get an error and my bltouch only starts clicking and then throws the error

Steven757, 2020-03-24:
Does anyone have final solution on this yet?

Steven23456, 2020-03-25:
> Whats the current status?
> I got my new v1.4 Turbo and i got everything to work except my Trianglelab BLTouch. Im currently on the newest Marlin 2.0.5
>
> If i disable Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN [...]

@xkuyax i'm not sure if 2.0.5.2 resolves the issue yet but I was able to get it to work using the changes recommended above in 2.0.5.1. The issue is the logic conflicts with itself ... so in Conditionals_LCD.h you need to change line 586 to read
`  #if Z_HOME_DIR < 0 //&& !HAS_CUSTOM_PROBE_PIN `

Then commenting out
`// Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`
in Configuration.h and although i'm not sure its needed (just in case) set the PIN
`#define Z_MIN_PROBE_PIN P0_10`

This worked for me.

LiS92, 2020-03-27:
> @xkuyax i'm not sure if 2.0.5.2 resolves the issue yet but I was able to get it to work using the changes recommended above in 2.0.5.1. The issue is the logic conflicts with itself ... so in [...]

Thanks ! That worked for me too. Marlin 2.0.5.2

djessup, 2020-03-28:
I've just been setting up my Ender 3 Pro with an BTT SKR 1.4 Turbo fitted with TMC2209's. My goal was to have sensorless homing + have the BLTouch used for UBL. 

Initially I had some issue with the z-homing but after experimenting I've been able to get it working well using the SKR1.4's dedicated z-probe pins. Here are the key configs I've landed on:

**Configuration.h**
```//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN   // i.e. comment this out
#define Z_MIN_PROBE_PIN P0_10

#define BLTOUCH

#define NOZZLE_TO_PROBE_OFFSET { 38, 0, 0 }     // set yours according to your printer
```

**Configuration_adv.h**
```#define BLTOUCH_SET_5V_MODE

#define SENSORLESS_HOMING

#define X_STALL_SENSITIVITY  100
#define Y_STALL_SENSITIVITY  100
#define Z_STALL_SENSITIVITY  100    // these are the values that work for me, but you may need to fine-tune
```

On the hardware side, BLTouch is wired up to the corresponding SERVOS/Probe pins on the motherboard. **I also have the original z-endstop microswitch still wired up** and this is critical to it working. Without this homing won't work for the z-axis. 

With this setup you can auto-home fine, and G29 etc to use the BLTouch for leveling. If you insist on removing the z-endstop switch you'll likely either need to swap the BLTouch z-pins to the normal z-endstop, or use SENSORLESS_PROBING and rely on stallguard for the z-axis. (Or hack Marlin like mentioned above, however this will make upgrading to newer versions harder so I'd avoid it unless absolutely necessary).

thelittlemike, 2020-03-28:
> 
> 
> I've just been setting up my Ender 3 Pro with an BTT SKR 1.4 Turbo fitted with TMC2209's. My goal was to have sensorless homing + have the BLTouch used for UBL.
> 
> Initially I had some issue with the z-homing but after experimenting I've been able to get it working well using the SKR1.4's dedicated z-probe pins. Here are the key configs I've landed on:
> 
> **Configuration.h**
> 
> ```
> #define Z_MIN_PROBE_PIN P0_10
> 
> #define BLTOUCH
> 
> #define NOZZLE_TO_PROBE_OFFSET { 38, 0, 0 }     // set yours according to your printer
> ```
> 
> **Configuration_adv.h**
> 
> ```
> 
> #define SENSORLESS_HOMING
> 
> #define X_STALL_SENSITIVITY  100
> #define Y_STALL_SENSITIVITY  100
> #define Z_STALL_SENSITIVITY  100    // these are the values that work for me, but you may need to fine-tune
> ```
> 
> On the hardware side, BLTouch is wired up to the corresponding SERVOS/Probe pins on the motherboard. **I also have the original z-endstop microswitch still wired up** and this is critical to it working. Without this homing won't work for the z-axis.
> 
> With this setup you can auto-home fine, and G29 etc to use the BLTouch for leveling. If you insist on removing the z-endstop switch you'll likely either need to swap the BLTouch z-pins to the normal z-endstop, or use SENSORLESS_PROBING and rely on stallguard for the z-axis. (Or hack Marlin like mentioned above, however this will make upgrading to newer versions harder so I'd avoid it unless absolutely necessary).

You do not need to have the endstops wired up. I'm not sure why people aren't able to get it working. I'm not a Marlin guru, so unfortunately, I can't really tell what's wrong. I just know that I have sensorless homing, stallguard, BLTouch 3.1 (authentic) connected to the dedicated pins, cooling fan on auto as well as hotend fan. Auto Home, bed leveling, all that stuff works for me. Here's my configs if maybe some of you guys can reverse engineer what I did.
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/4396550/Configuration.zip)

djessup, 2020-03-28:
@thelittlemike I'll take a look and see if I can figure it out, thanks for posting your configs. Out of interest, what version of Marlin are you using, and did you start with the BTT fork or straight from the official repo? (For this built I started from the BTT fork, but I plan to switch back to the mainstream repo shortly - fork is 2.0.1.)

djessup, 2020-03-28:
@thelittlemike PS - Love the custom machine name ;)

thelittlemike, 2020-03-28:
> 
> 
> @thelittlemike PS - Love the custom machine name ;)

Haha, thanks. I started directly with Marlin 2.0.4.4, then started over from scratch with each revision since, because I don't know how to upgrade. Lol.

So the configs I sent you are for 2.0.5.2 that I set up yesterday. I basically just opened up my config and advanced config file in Visual Studio and copied my changes (I generally put // EDIT HERE for the major edits that might break something).

Really that simple.

Oh, as far as wiring - so the hotend fan is plugged into HE1 and the part cooling fan is in FAN0 if that helps.

MaximusStalsky, 2020-03-28:
Hey folks! Im stuck with same issue( 
None of your suggestions helped.

I have a skr 1.4+btt24tft and a 3dtouch from TringleLabs.
Wiring looks ok. 
Not from TFT menu, or from FullGraphicDisplay menu or G28 i can't make 3dtouch release the probe. 
When i turn on the printer 3dtouch does self test and the probe works fine. 
I use platformIO.

about #if Z_HOME_DIR < 0 //&& !HAS_CUSTOM_PROBE_PIN
when i change it it loks like the part after // is now a comment not a part of the function. it's supposed to be that way?

I'm not so skilled yet. Can any one check my cfg plz? It compiles without errors but there has to be one...

[Configuration.h.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/4397208/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/4397209/Configuration_adv.h.txt)
[Conditionals_LCD.h.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/4397210/Conditionals_LCD.h.txt)
[pins_BTT_SKR_V1_4.h.txt](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/4397211/pins_BTT_SKR_V1_4.h.txt)

arminbl, 2020-03-30:
> @xkuyax i'm not sure if 2.0.5.2 resolves the issue yet but I was able to get it to work using the changes recommended above in 2.0.5.1. The issue is the logic conflicts with itself ... so in [...]

The issue is not resolved in 2.0.5.2 but this worked for me too with 2.0.5.2. 
The line #define Z_MIN_PROBE_PIN P0_10 is not nescessary in configuration.h, because it is already defined in pins_BTT_SKR_V1_4.h:
//
// Z Probe (when not Z_MIN_PIN)
//
#ifndef Z_MIN_PROBE_PIN
  #define Z_MIN_PROBE_PIN                  P0_10
#endif

djessup, 2020-03-31:
@thelittlemike I compared your config with mine, and they were very similar regarding z-probing with the exception of your modification to the board pins for `Z_STOP_PIN` and `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` being disabled per default, where I had it enabled. 

The latter I'd enabled after failing to get the BLTouch to z-home, but even before that when homing Z the probe did not get deployed and Z would home like normal (with just an endstop switch).

I have seen several videos on using BLTouch + SKR 1.4 which say they had to use the normal Z-stop port, which makes sense since that would match the default `Z_STOP_PIN` setting in Marlin. Looking deeper in the code Marlin will only use the servo if `HOMING_WITH_Z_PROBE` is true, but this will never be the case if `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` is enabled:
```
  #if DISABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)
    #define HAS_CUSTOM_PROBE_PIN 1
  #endif
  #if Z_HOME_DIR < 0 && !HAS_CUSTOM_PROBE_PIN
    #define HOMING_Z_WITH_PROBE 1
  #endif
```

I can kind of see the logic here, but it makes the assumption that the z-min pin will only ever be used by a physical endstop rather than a probe and I didn't see this clearly stated anywhere during my searches. It also explains why others have suggested removing `&& !HAS_CUSTOM_PROBE_PIN` as a workaround.

For now my preference is to conditionally set the `Z_STOP_PIN` in the BTT1.4 pins file based on whether `BLTOUCH` is enabled, but this is a bit of a hack. I think some broader refactoring is needed to make this more rational.

**tl;dr**
Make sure the `Z_STOP_PIN` is set to `P0_10` on the SKR 1.4 (the dedicated BLTouch z-stop pin) OR use the normal Z-stop port on the board.
**/tl;dr**

thelittlemike, 2020-03-31:
> @thelittlemike I compared your config with mine, and they were very similar regarding z-probing with the exception of your modification to the board pins for `Z_STOP_PIN` and [...]

It's definitely a bit of a hack, but the idea behind it is so that you can use both physical endstops as well as software endstops. Not sure how much refactoring could be done without forcing one or the other. But anyway, my method works, so yay!

djessup, 2020-04-01:
The part that doesn't make sense to me is that the logic in Marlin is "if the probe uses a custom pin it can't be used for z-homing". Having a discussion w/ Marlin devs over here if you're interested in following - https://github.com/MarlinFirmware/Marlin/pull/17063

thelittlemike, 2020-04-01:
> 
> 
> The part that doesn't make sense to me is that the logic in Marlin is "if the probe uses a custom pin it can't be used for z-homing". Having a discussion w/ Marlin devs over here if you're interested in following - [MarlinFirmware/Marlin#17063](https://github.com/MarlinFirmware/Marlin/pull/17063)

Oh gosh, Chris is flipping out in that thread. I'm not touching it with a 10 foot pole. Lol

djessup, 2020-04-01:
> Oh gosh, Chris is flipping out in that thread. I'm not touching it with a 10 foot pole. Lol

I can understand why. The original "fix" was a mess and there was no sense behind it aside from "it works for my specific situation". I've raised a new PR I think it a little more sane and shouldn't have any backwards compatibility issues.

https://github.com/MarlinFirmware/Marlin/pull/17359

PapiOrso, 2020-04-01:
Hi, I am attaching my configuration files for comparison. On my SKR 1.4 the Bltouch works well connected to the dedicated Probe connector.

[Configurations.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/4414257/Configurations.zip)

thelittlemike, 2020-04-01:
> 
> 
> > Oh gosh, Chris is flipping out in that thread. I'm not touching it with a 10 foot pole. Lol
> 
> I can understand why. The original "fix" was a mess and there was no sense behind it aside from "it works for my specific situation". I've raised a new PR I think it a little more sane and shouldn't have any backwards compatibility issues.
> 
> [MarlinFirmware/Marlin#17359](https://github.com/MarlinFirmware/Marlin/pull/17359)

Agreed. Though my fix of changing the pin assignment was hacky, it was dead simple and worked. So I'm not sure why all the other changes. In either case, hopefully a solution comes up that will allow for both mechanical endstops as well as software-based, and retaining legacy support.

djessup, 2020-04-01:
> Hi, I am attaching my configuration files for comparison. On my SKR 1.4 the Bltouch works well connected to the dedicated Probe connector.

@PapiOrso Do you have the BLTouch z-cable plugged into the standard z-endstop port? Otherwise I can't see how this would work unless you've also modified the pins file to change `Z_MIN_STOP` to `P0_10`?

PapiOrso, 2020-04-02:
@djessup It's strange I know but it's the only way that makes the BlTouch connected to the Probe connector working. I discovered it casually after doing tests on tests. I haven't made any changes to the PINs. I forgot to say that in the standard z-endstop connector the switch cable must remain connected otherwise it will not work.

Steven23456, 2020-04-02:
I did a little bit of digging into the code (I did not do any tracing) .. based on what I see 
`Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`
should NOT be defined if the BLTouch is connected to the provided port on the SKR 1.4 board connected to pin 0.10 I'm confident that this is a safe assumption based on the comments in the configuration.h file as well as looking at how the symbol is used in the code (specifically BLTouch.cpp)

Here is what I mean: in BLTouch.cpp lines 93-101 (Marlin 2.0.5.2)

```
bool BLTouch::triggered() {
  return (
    #if ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)
      READ(Z_MIN_PIN) != Z_MIN_ENDSTOP_INVERTING
    #else
      READ(Z_MIN_PROBE_PIN) != Z_MIN_PROBE_ENDSTOP_INVERTING
    #endif
  );
}
```

you can see that a "triggered" state for the probe uses the Z_MIN_PIN unless Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN is disabled.

However, if you do disable Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN .. then in Conditionals_LCD.h never sets HOMING_Z_WITH_PROBE 1

see here from Conditionals_LCD.h lines 575 - 605

```
#if HAS_BED_PROBE || EITHER(PROBE_MANUALLY, MESH_BED_LEVELING)
  #define PROBE_SELECTED 1
#endif

#if HAS_BED_PROBE
  #if DISABLED(NOZZLE_AS_PROBE)
    #define HAS_PROBE_XY_OFFSET 1
  #endif
  #if DISABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)
    #define HAS_CUSTOM_PROBE_PIN 1
  #endif
  #if Z_HOME_DIR < 0 && !HAS_CUSTOM_PROBE_PIN 
    #define HOMING_Z_WITH_PROBE 1
  #endif
  #ifndef Z_PROBE_LOW_POINT
    #define Z_PROBE_LOW_POINT -5
  #endif
  #if ENABLED(Z_PROBE_ALLEN_KEY)
    #define PROBE_TRIGGERED_WHEN_STOWED_TEST 1 // Extra test for Allen Key Probe
  #endif
  #if MULTIPLE_PROBING > 1
    #if EXTRA_PROBING
      #define TOTAL_PROBING (MULTIPLE_PROBING + EXTRA_PROBING)
    #else
      #define TOTAL_PROBING MULTIPLE_PROBING
    #endif
  #endif
#else
  // Clear probe pin settings when no probe is selected
  #undef Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#endif
```
HOMING_Z_WITH_PROBE 1 is used by G28.cpp to home the Z axis (when a probe is connected)

I can certainly see a situation where you have both a probe connected to the 0.10 pin and a switch connected to the Z Stop pin 1.27 ... working .. but if you have Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN defined .. you are basically using the Z Stop switch to home the printer ...

meltonpieman, 2020-04-02:
> So I got the BLTouch (3.1 version) to work just fine making no edits whatsoever to vanilla Marlin other than enabling BL Touch and 5V and a single change to the pins file:
>
> #else
> #define Z_STOP_PIN [...]

Quite honestly, I am getting confused with all this. I have an SKR 1.4 with TMC2209s and a BLTouch 3.0. I am setting the board and Marlin up before placing it into my Ender 3. 
I get the sensorless homing on X & Y but with your pin assignment to the BLTouch plug (black and white) does that mean sensorless homing works on Z as well? 
Or, do I have to remove the pin on the Z motor TMC2209  - as seem to have to if I plug the BLTouch into the Z endstop socket?
Sorry to seem a bit dim - but I probably am with all this isolation in the UK!
David

djessup, 2020-04-02:
@PapiOrso That makes sense for the reasons @Steven23456 has detailed. 

Basically in marlin you either have to modify the pin definitions to change the z-endstop to the SKR 1.4 BLTouch end stop pin (P0_10) or plug the BLTouch has into the normal z-endstop port. 

You can have a custom pin for z-min and use it for homing without modifying Marlin. 

I have a PR open to address this:
https://github.com/MarlinFirmware/Marlin/pull/17359

djessup, 2020-04-02:
> Quite honestly, I am getting confused with all this. I have an SKR 1.4 with TMC2209s and a BLTouch 3.0. I am setting the board and Marlin up before placing it into my Ender 3.
> I get the sensorless [...]

@meltonpieman I have a very similar setup, Ender 3 Pro + BLTouch 3.1 + 2209s. Sensor less homing on X/Y and BLTouch for Z. Quickest way to get it working is to change Z_MIN_PIN to P0_10, or you can try my fix from the PR linked above.

Steven757, 2020-04-02:
@djessup you’re correct, because you’re re assigning the z stop pin. I can see how that would work just fine. At the end of the day it’s a minor bug..l but bug it is still 🤗

Kamoool, 2020-04-03:
Hello,

I've got SKR V1.4 Turbo and 3DTouch V2.0 from Makerbase, this one:
https://pl.aliexpress.com/item/32890485972.html?spm=a2g0s.9042311.0.0.66565c0fcPeQiG

I thing I've tried almost everything to make it work, but it wont. With M119 z_probe is open when deployed, and TRIGGERED when not.

Can anyone send me working configs?
Im using TMC2208's.

spamplin, 2020-04-09:
Finally got mine working with physical endstops, BLTouch and Bilinear leveling. Latest main release 2.0.5 I think - change to the PINS.h file + define BLTouch and 5v mode in config as described above. Very pleased.

Ratler5, 2020-04-10:
Hey Guys, I just got my SKR1.4 Turbo set up in my Tevo little monster(delta). I can not get my BLTouch to work at all. I see most of these instructions are for Ender 3 and other similar printers. What do i need to setup for my Delta since it needs the Z end stop for homing and the BLTouch for bed leveling?

sputnik378, 2020-04-12:
I've been messing with this for days, and I keep ending up with a nonworking firmware. I just want someone to push the EASY button on this for me... I am apparently not talented enough to sort through this and figure out how to make it work for my Ender 3 with SMK 1.4 Turbo & TMC2208 and BLTouch.

Steven757, 2020-04-12:
> I've been messing with this for days, and I keep ending up with a nonworking firmware. I just want someone to push the EASY button on this for me... I am apparently not talented enough to sort through this and figure out how to make it work for my Ender 3 with SMK 1.4 Turbo & TMC2208 and BLTouch.

What type of printer? And what mb exactly?... the SKR?

sputnik378, 2020-04-12:
> What type of printer? And what mb exactly?... the SKR?

Ender 3, SKR 1.4 Turbo with TMC2208 Stepper Drivers, and the BLTouch

I also have Octoprint on a RPi 3B+ and the TFT 3.5 v2 touchscreen

Steven23456, 2020-04-12:
DM me your email on Twitter @stevenallen , let’s connect that way... I have almost the same setup. But we need to make sure your wiring is correct

sputnik378, 2020-04-12:
> DM me your email on Twitter @StevenAllen , let’s connect that way... I have almost the same setup. But we need to make sure your wiring is correct

Your twitter won't accept a DM. @sputnik378  is mine.

Culem94, 2020-04-13:
hello guys!

I have SKR 1.4 turbo with tmc 2209 and marlin 2.0.5.3...my bltouch probe do not deploy when z homing...I tried all configurations that I found here...It is not working for me...can you help me?

djessup, 2020-04-17:
@Culem94 Check you wiring is correct. Use GCode or menus in Marlin to manually test/deploy/stow to confirm.

FYI I originally had a cheap clone that wouldn't work no matter what firmware/hardware changes I tired. I suspect it was DOA. Got an official BLTouch after that and it worked fine. Not saying all clones are bad or won't work, but there are certainly some out there with issues.

japri, 2020-04-17:
> So I got the BLTouch (3.1 version) to work just fine making no edits whatsoever to vanilla Marlin other than enabling BL Touch and 5V and a single change to the pins file:
>
> #else
> #define Z_STOP_PIN [...]

Hi mike did you cut pin on the 2209?

thelittlemike, 2020-04-17:
> Hi mike did you cut pin on the 2209?

No, I did not. You should have no need to cut the DIAG pin.

PainHealer, 2020-04-18:
> @Culem94 Check you wiring is correct. Use GCode or menus in Marlin to manually test/deploy/stow to confirm.
>
> FYI I originally had a cheap clone that wouldn't work no matter what firmware/hardware [...]

I have the same problem as @Culem94. My BLTouch is a clone but it worked absolutely flawless ever bevore on a GT2560 board. I have attached it to the dedicated Probe connector(s), checked the wiring, the red led is lit permanently and I can deploy/stow/test it with M280. The problem I have is that i simply does not deploy when starting to home Z. I do not have a physical switch connect to the ZU-Min connector. Any idea what I can do to make it deploy?

[104 more replies]