bigtreetech/BIGTREETECH-SKR-V1.3 #208: SKR v 1.3 not showing on windows10

psydon6, 2020-01-27:
2 major issues, which are leading me to believe the board is dead.  Yes the power light is on, however, I cannot flash the .bin onto the board and second, the board is not showing up on my pc.  Have tried multiple cables, multiple sd cards.  Reset button and power cycling both provide no results either.

damienj210, 2020-01-27:
I had this problem also on a computer that previously could connect. Connected to a different computer and it worked fine. Somehow the drivers got corrupted in the first computer. Still havent been able to fix them.

psydon6, 2020-01-27:
So, I thought of doing that exact thing, however, still nothing, brought in
a different computer, but still no fix.

Cerv4ntez, 2020-01-29:
re-format the sd card and put only the bin file on it before putting it back in, this fixes it most of the time, worked for me a while ago

psydon6, 2020-01-30:
Tried that, no luck.  I tried a few other things, won't boot via usb.  I'm
gonna try RMA'ing the board.  Ive seen online a few people have received
DOA boards like this.

Cerv4ntez, 2020-01-31:
maybe try replacing the sd card before you do

psydon6, 2020-02-01:
Already did with 3 different sd's

Marsman1970, 2020-02-02:
If you plug in the USB while windows in on does it see it at all ? does it says device not recognized or not even see it at all.
If it doesn't show up at all the firmware might be missing not there at all.

psydon6, 2020-02-03:
It doesn't show up at all.  I tried flashing it, but no luck.

Marsman1970, 2020-02-03:
With the boot loader i uploaded the others don't work
https://github.com/Marsman1970/SKR-V1.3-Bootloader
Michael

harpagophytum, 2020-10-18:
take a look here

Cordialement,