bigtreetech/BIGTREETECH-SKR-V1.3 #209: Lerdge TMC2209 UART not working

Narolakk, 2020-01-28:
I don't get the UART mode to work.  I tried different things.  
I jumpered the TX / RX pins with a jumper and I soldered down two pins that go into the socket.  always get connection error with M122.

I have not yet tried the red pins for bridges.

[image]
[image]

Narolakk, 2020-01-28:
UART works!  you have to solder two pads together on the bottom.
[image]

closed 2020-01-28 by Narolakk

PaedeG, 2020-03-05:
I've got exactly the same problem here..
I soldered as you mentioned..  
have you done only the soldering to get it work? or are you using the jumper (rx/tx) too?
Thanks a lot

[image]

phiphe, 2020-05-06:
I have to put this 2209 v1.0 in skr.1.3... can you help.me?.
I must to insert in down side 2 pins and sold this?.
Thanks and regards.

mgpsantos, 2020-10-31:
Hi
Did you solve the lerdge tmc2209 driver problem in uart mode?

derTimm0, 2022-01-08:
i know, old tread ;-) but i has the same problem. can some one help me? ;-)