bigtreetech/BIGTREETECH-SKR-V1.3 #212: Pinout for SKR V1.3 AUX-1

goldjunge91, 2020-01-29:
hello guys
i need the PIN numbers for the AUX-1 pins. I will connect a PT100 to it
the red marked Pin numbers i need.  
I found the other pictures on google. but it doesn't work
[image]
[image]
[image]

bdelia, 2020-04-17:
@goldjunge91 , did you ever figure this out?  I am having issues getting the board to post when the firmware is configured to use a PT100 amp board.

Per instructions I have, I set TEMP_SENSOR_0 to 20 in Configuration.h, but the board does not post configured that way. If I change it to 1 then it posts but the PT100 board is not configured. What am I missing? All else seems to be working.

I feel like I am missing something obvious. Any help is greatly appreciated.

goldjunge91, 2020-04-17:
yes you need to change in pins btt_skr_common.h 
#ifndef TEMP_0_PIN
  #define TEMP_0_PIN             --->       P0_24_A1  // A1 (T1) - (68) - TEMP_0_PIN
but i have never found a pin layout for a analog input pin. 
when you know how is the number for the AUX1 TX0 pin then u can change  #define TEMP_0_PIN or 
u can test with downgrade the voltage to 3.3V and use the normal temp pin
i have shipped back the board and now i use a mks board

bdelia, 2020-04-19:
If anyone needs further help on this issue I finally had someone point me to this reference manual for the LPC1769/68/67/66/65/64/63 micro-controller, very helpful.  Page 10 is where the pin assignments get defined.

https://www.nxp.com/docs/en/data-sheet/LPC1769_68_67_66_65_64_63.pdf?fbclid=IwAR3QKfZhRvsdNU1BDoAlD6861QcTgNXTYBp2D8TwN1QkZDdlmCYKbTKfDlY

goldjunge91, 2020-04-19:
I will test it in 1 week and while Update what work :-) thanks men

bdelia, 2020-04-19:
@goldjunge91, I ended up using #define TEMP_0_PIN P0_03_A6

goldjunge91, 2020-04-20:
Can you upload a picture what pin this is ?

bdelia, 2020-04-20:
> Can you upload a picture what pin this is ?

It is the one labeled RX0 within the AUX-1 header.  Let me know if you still need a picture and I can add one later today.

brazda, 2021-08-19:
I just wanted to add the Klipper configuration as I just spent days looking for it and could not find the answer. 
For klipper and PT100 and skr v1.3  

sensor_type: PT100 INA826
sensor_pin: P0.2

It needs to be plugged into the 3.3V