bigtreetech/BIGTREETECH-SKR-V1.3 #214: SKR 1.4 Turbo | Blank LCD

77katz77, 2020-01-30:
I just finished configuring my BigTreeTech SKR 1.4 Turbo using the provided modifications and configured Marlin 2.0 config.[https://www.dropbox.com/s/ppjfflhf3j5yzh2/MarlinV2.0%20SKRV1.1%20instruction.docx?dl=0](url)
I was able to load the compiled .bin file to the SKR through the SD Card. However, whenever I power the board on, the LCD screen remains blank (but lit). I know the screen works and I don't think anything was modified in the config that would effect the display. I've also tried powering off of the usb and 12v power supply, both result in the same problem. If anyone has any idea what the issue is please let me know.

(Config and Platformio files)
https://github.com/77katz77/SKR1.4-Turbo.git

Cerv4ntez, 2020-01-30:
in your config you defined a #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

ist that also what you are really using?

77katz77, 2020-01-31:
> in your config you defined a #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
> 
> ist that also what you are really using?

So I was able to define ZONESTAR_LCD (what's used on the anet a8). The problem is now I get an error "'ACD_KEYPAD_PIN' was not declared in this scope" (in temperature.cpp) whenever I try and compile. From what I understand this has to do with how the buttons on the display are assigned. I've seen multiple threads saying to reassign some of the UI_DISPLAY pins. If this is the case how would I achieve this (nothing in Config.h and there doesn't seem to be a Display.h file as others have suggested)?

Cerv4ntez, 2020-01-31:
i would start from example config files for your printer and modify anything concerning the new board much easier than the other way around.

ellensp, 2020-02-01:
Its ADC_KEYPAD_PIN not ACD_KEYPAD_PIN  
If just plugged in would go to pin 1.20 but 1.20 is not analog. So this will not work
You would need a special cable to put the analog wire on a unused analog input. 
You also have to make sure not to exceed 3.3v, as analog pins are not 5v tolerant.  
only these pins are defined as analog...  
#define P0_02_A7 P0_02
#define P0_03_A6 P0_03
#define P0_23_A0 P0_23
#define P0_24_A1 P0_24
#define P0_25_A2 P0_25
#define P0_26_A3 P0_26
#define P1_30_A4 P1_30
#define P1_31_A5 P1_31

Your original 12864 screen probably has its cables rotated 180 degrees from the skr1.4. this is very  common , depends on where the screen came from..   I pull off the plastic kyeds sockets on the back of the lcd and plug the cables in backwards, that way they are correct on the controller end.

77katz77, 2020-02-04:
> Its ADC_KEYPAD_PIN not ACD_KEYPAD_PIN
> If just plugged in would go to pin 1.20 but 1.20 is not analog. So this will not work
> You would need a special cable to put the analog wire on a unused analog [...]

Thanks for this detailed explanation. I decided to just buy a TFT 3.5 which I was able to get working pretty easily. If I get the time I'll make sure to come back and hopefully will be able to get the zonestar working.

closed 2020-02-04 by 77katz77