bigtreetech/BIGTREETECH-SKR-V1.3 #215: BTT TMC2130SPI V3

FacusFPV, 2020-01-31:
good morning

I have a skr1.3 working with tmc2208 uart. I bought some btt tmc2130 spi v3 but I can't make the engines move.

TMC2130 V3.0 SPI Mode
https://www.aliexpress.com/item/32970150483.html?spm=a2g0s.9042311.0.0.56744c4diNkGvn

SW SPI enabled
CFG 4 and CFG 5 to gnd
Spi without resitor

X 0x00: 00: 00: 00 Bad answer!
And 0x00: 00: 00: 00 Bad answer!
Z 0x00: 00: 00: 00 Bad answer!
E 0x00: 00: 00: 00 Bad answer!
Connection test X ... Error: all LOW
Connection test Y ... Error: All LOW
Connection test Z ... Error: All LOW
Connection test E ... Error: All LOW

Talha909, 2020-01-31:
Have you give the power to the board. When you only power from the usb then its not work

FacusFPV, 2020-01-31:
the printer has a 12v source connected

XzenHD, 2020-02-06:
I have the same problem. Printer is connected to 12v power-source, with the jumper in the right possition on the motherboard. I have stopped trying with the BTT stepper-drivers after having tried 8 different peaces

Talha909, 2020-02-12:
> I have the same problem. Printer is connected to 12v power-source, with the jumper in the right possition on the motherboard. I have stopped trying with the BTT stepper-drivers after having tried 8 different peaces

which tmc do you have? Please paste the config files

Talha909, 2020-02-12:
> the printer has a 12v source connected

only one tmc is ok other 3 are not. Please past the pic and config files.