bigtreetech/BIGTREETECH-SKR-V1.3 #217: no BTT SKR V1.4 schematic

ellensp, 2020-02-01:
I wanted to check the endstops pullup resistor values, but no schematics

Please upload schematics

Marsman1970, 2020-02-03:
I would say it would be have the same resistors as the Skr v1.3 they are 10k. If you look at the end-stop connectors on the left there is a row  of resistors going down if there are resistors that say 103 they are 10k. I tried to look at the picture of the 1.4 board but it's not clear enough for me to see. Are you having problems with your end-stops.
Michael

ellensp, 2020-02-03:
You would be wrong... mini ir probes don't work on skr 1.4 endstop pins and they do work on skr 1.3 endstop pins and we know its a different pullup on the board. Just wanted to confirm what value.

Marsman1970, 2020-02-03:
Now you got me interested i had a look at Skr v1.1 and v1.3  and i found that the   v1.1 the limit switch socket and  resistor is connected to 3.3 volt rail and used a 4.7k pull-up resistor but on the v1.3 the limit switch socket   is connected to 5 volt rail and used a 10k pull-up that is connected to the 3.3 volt rail. I wonder what the changed on the 1.4  according to the diagram  the limit switch socket  is still  connected to 5 volt rail i can almost see the a 10k resistor but the image is just not clear enough.
Michael

DeviousPenguin, 2020-02-06:
Schematics or even the components used would be good to have.

I'd quite like to safely run my motors at a higher voltage than 24v, but with no schematics or component list I have no idea how safe this is!

rrichX, 2020-02-09:
I raised a question over end stop behaviour last week in the Marlin Support Discord channel.

I went from a SKR 1.3 > 1.4 and my IR Probe failed to work - being reported as Trigger, reading around I decided to give the probe pin a try and now it works - doing an M48 it's far more accurate than it ever was on the 1.3.

I did break out my magnifier and as far as I can, the values have changed of the resistors but I'm also curious to why the probe pin works so much better. I thought I had read about changes to the end stops to help optical end stops, I wonder if my issue is a result of changes.

asdmin, 2020-03-09:
dup to https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/184

closed 2020-03-10 by ellensp