bigtreetech/BIGTREETECH-SKR-V1.3 #221: ESP-01S connection to wifi connector on the skr-1.4-turbo

malexzx, 2020-02-04:
I try to connect the ESP-01S module to the SKR-1.4-turbo Wi-Fi connector on the board, but to no avail. 
But connection to TFT connector on the board works well (It has RX,TX, RST and only 3.3v I get from different place). While studying the pins P1_28, P1_29, so marked as wifi on the board schema, I get it not found in the provided Marlin here.
Does who have success, on the connection this module to board, on WiFi connector?

TJ1999, 2020-02-14:
Hi,
As the Pin scheme on the github page of the SKR1.4 shows are the RX and TX pins for the wifi module located at pins 4.29 and 4.28.
I also have an Wifi module now. I connected it to the skr1.4 turbo and set SERIAL_PORT_2 to 3  in configuration.h because the RX and TX pins belong to UART3 (have a look at the LPC1769 datasheet from NXP).  Strange is that there are also other pin configurations for UART3.
But I am facing problems, the Wifi Module works (internet page), but there is no connection to the printer.
Does anyone got the wifi modules to work?

malexzx, 2020-02-14:
Hi,
internet page works, because board powered by 3.3v

TJ1999, 2020-02-16:
I think I found the problem:
So I figured out that something with the UART Communication on UART3 was wrong. On UART0 (where the tft displays are normally connected to) it worked perfectly.

After some search I found the definitions of the UART pins for the LPC176x series.
I found out that not Marlin has a bug, but the arduino framework for the LPC176x series.

So to make UART3 (Wifi connector) working on the BTT 1.4 Turbo board, you have to go to:
<innstallation directory of platformio>/packages/framework-arduino-lpc176x/cores/arduino/HardwareSerial.h
At the function <begin> the pins are configured for UART. UART0 is configured correctly but on UART3 you have to make the following changes:
`} else if (UARTx == LPC_UART3) {
      // Initialize UART3 pin connect
      PinCfg.Funcnum = 3;
      PinCfg.OpenDrain = 0;
      PinCfg.Pinmode = 0;
      PinCfg.Pinnum = 28;
      PinCfg.Portnum = 4;
      PINSEL_ConfigPin(&PinCfg);
      PinCfg.Pinnum = 29;
      PINSEL_ConfigPin(&PinCfg);
}`
After rebuilding the project my ESP and my tft display work fine together.

I hope this was helpful!

icebat, 2020-02-17:
I followed TJ1999's instructions using platformio with an SKR 1.4 board and an ESP-01S plugged into the Wi-Fi connector on the board.

Changed /.platformio/packages/framework-arduino-lpc176x/cores/arduino/HardwareSerial.h
to the following:
} else if (UARTx == LPC_UART3) {
      // Initialize UART2 pin connect
      PinCfg.Funcnum = 3;    //was 1
      PinCfg.OpenDrain = 0;
      PinCfg.Pinmode = 0;
      PinCfg.Pinnum = 28;    //was 0
      PinCfg.Portnum = 4;    //was 0
      PINSEL_ConfigPin(&PinCfg);
      PinCfg.Pinnum = 29;   // was 1
      PINSEL_ConfigPin(&PinCfg);
    }
Changed configuration.h
to the following:
#define SERIAL_PORT -1
#define SERIAL_PORT_2 3

It works! I can use Octopi on the USB port and the ESP-01S plugged into the wifi connector at the same time! Thank you for the excellent work!

ehtnevets, 2020-03-29:
New to platformio but how do I rebuild the packages. I just did compile but it seems only compile the SKR project.

Thanks

TJ1999, 2020-03-29:
You should to go to the platformIO installation Folder, because in there are the arduino libraries. I use Linux so my path is 
home/tj/.platformIO/
I dont know how it is on Windows or Mac. 
In this Folder it should be sufficient to just edit the HardwareSerial.h Header file in the platformIO folder under 
packages/framework-arduino-lpc176x/cores/arduino/HardwareSerial.h

Normally the compiler recompiles if a file changed... So after changing the values of the UART pin configuration, the compiler should compile with the new values.
So there is no special need to do a full rebuild. You can just Build the Project as normal.

warener, 2020-04-16:
if I set SERIAL_PORT_2 to 3 I can no longer connect to the pc on the usbport. is that normal?

picard12, 2020-04-24:
I follow yours instruction, and all works, but now my tft35 v2 is no longer connect to the board.

xkongox, 2020-04-29:
Hi, I have SKR 1.4 mainboard with Merlin 2.0.5 + BT touch and REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER - I followed instructions with changing options in HardwareSerial.h.
I have tried install ESP-01S.  It works in programator I can login via WIFI.  But when install to the board, sth goes wrong. Sympthoms. 
1. Normally during booting LCD buzzer is loud and annoying.  With WIFI it is rather silent, so I suspect it is some power problem or short circuit.
2. Menu is jumping from Info screen to main screen. I hear clicking and it is loop. I can't stop it. 
3. ESP-01S within 10-15 seconds is extremely hot. 

When I remove module, all works fine.  When I change HardwareSerial definition to default, the problem is still excists.

Maybe somone can say what is relation between LCD and WIFI on SKR 1.4? I work on 12V power.

TJ1999, 2020-04-30:
> if I set SERIAL_PORT_2 to 3 I can no longer connect to the pc on the usbport. is that normal?

Yes this is normal. At leaast on my Sidewinder X1. Before the ESP module SERIAL_PORT_2 was -1 which stands for USB communication. So you have to decide whether you want USB connection or a wifi connection.

TJ1999, 2020-04-30:
> I follow yours instruction, and all works, but now my tft35 v2 is no longer connect to the board.

Maybe you changed the Serial_Port number which belongs to your display?

TJ1999, 2020-04-30:
> Maybe somone can say what is relation between LCD and WIFI on SKR 1.4? I work on 12V power.

Well, double check your connections. As I got my Wifi module I almost connected it the wrong way because if I would correctly connect it, it would face away from the board. I had to use jumper wires.
That could be the reason why the buzzer is not so loud and the menu is flickering and also the ESP runs hot. Microcontrollers don't like Negative Voltage Supply.

warener, 2020-05-02:
can i add SERIAL_PORT_3 to marlin to work lcd wifi and pc

> Yes this is normal. At leaast on my Sidewinder X1. Before the ESP module SERIAL_PORT_2 was -1 which stands for USB communication. So you have to decide whether you want USB connection or a wifi [...]
can i add SERIAL_PORT_3 to marlin to work lcd wifi and pc

picard12, 2020-05-21:
> > I follow yours instruction, and all works, but now my tft35 v2 is no longer connect to the board.
> 
> Maybe you changed the Serial_Port number which belongs to your display?
How i can find the right serial_port for the tft and display?

picard12, 2020-05-21:
I found an issue, the tft 35v2 share the serial_port with the esp01s. How i can change the serial_port of tft?

brotherchris, 2020-05-27:
Is there any way to make these changes in the platformio.ini file instead?

PinCfg.Funcnum = 3; //was 1
PinCfg.OpenDrain = 0;
PinCfg.Pinmode = 0;
PinCfg.Pinnum = 28; //was 0
PinCfg.Portnum = 4; //was 0
PINSEL_ConfigPin(&PinCfg);
PinCfg.Pinnum = 29; // was 1
PINSEL_ConfigPin(&PinCfg);

Drzet, 2020-08-24:
Trying to figure out, did any of you manage to make ESP2866 work plugged into BOARD, not TFT ?
I only get a single blue blink on power-up, nothing else.
I studied the official pinout pdf [here](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.4/Hardware/BTT%20SKR%20V1.4PIN.pdf) and found that apart from ground , 3.3V, tx and rx all other pins are N/C !? 
Am I missing something?

mattster98, 2020-08-25:
I have moved on from this board, but did find that I needed to solder a jumper to enable the ESP in my case.  Might be the same and you can easily test with a hand-held jumper or other temporary short.  Not sure if that's the issue or not, but worth checking.

https://github.com/luc-github/ESP3D/issues/476#issuecomment-641558747

Drzet, 2020-09-02:
> Trying to figure out, did any of you manage to make ESP2866 work plugged into BOARD, not TFT ?
> I only get a single blue blink on power-up, nothing else.
> I studied the official pinout pdf [...]

It works. Royal pain in the butt that SKR

jamon, 2020-09-27:
for anyone trying to do this recently, instead of modifying your .platformio files, now you just need to add an extra define to your Configuration.h or somewhere else, at your discretion:

#define LPC_PINCFG_UART3_P4_28

picard12, 2020-10-04:
> #define LPC_PINCFG_UART3_P4_28

Hei Jamon, you are a genius. the last platformio update has broken the previuos set up.

Drzet, 2020-10-04:
In 2.0.7 and bugfix this definition already exists in Pins_BTT_SKRv1_4.h
Adding #define LPC_PINCFG_UART3_P4_28 to Configuration.h produces warning: "LPC_PINCFG_UART3_P4_28" redefined
  113 | #define LPC_PINCFG_UART3_P4_28
Anyways, with or without additionally defining UART3 it's **broken again.**

xstefa12, 2020-10-13:
> 
> 
> In 2.0.7 and bugfix this definition already exists in Pins_BTT_SKRv1_4.h
> Adding #define LPC_PINCFG_UART3_P4_28 to Configuration.h produces warning: "LPC_PINCFG_UART3_P4_28" redefined
> 113 | #define LPC_PINCFG_UART3_P4_28
> Anyways, with or without additionally defining UART3 it's **broken again.**

For me \.platformio\packages\framework-arduino-lpc176x\cores\arduino\HardwareSerial.h looked:
 
else if (UARTx == LPC_UART3) {
      // Initialize UART3 pin connect
      PinCfg.OpenDrain = 0;
      PinCfg.Pinmode = 0;

      #if defined(LPC_PINCFG_UART3_P4_28)
        PinCfg.Funcnum = 3;
        PinCfg.Portnum = 4;
        PinCfg.Pinnum = 28;
        PINSEL_ConfigPin(&PinCfg);
        PinCfg.Pinnum = 29;
        PINSEL_ConfigPin(&PinCfg);
      #elif defined(LPC_PINCFG_UART3_P0_25)
        PinCfg.Funcnum = 3;
        PinCfg.Portnum = 0;
        PinCfg.Pinnum = 25;
        PINSEL_ConfigPin(&PinCfg);
        PinCfg.Pinnum = 26;
        PINSEL_ConfigPin(&PinCfg);
      #else
        PinCfg.Funcnum = 2;
        PinCfg.Portnum = 0;
        PinCfg.Pinnum = 0;
        PINSEL_ConfigPin(&PinCfg);
        PinCfg.Pinnum = 1;
        PINSEL_ConfigPin(&PinCfg);
      #endif

but have to look:

else if (UARTx == LPC_UART3) {
      // Initialize UART3 pin connect
      PinCfg.OpenDrain = 0;
      PinCfg.Pinmode = 0;

      #if defined(LPC_PINCFG_UART3_P4_28)
        PinCfg.Funcnum = 3;
        PinCfg.Portnum = 4;
        PinCfg.Pinnum = 28;
        PINSEL_ConfigPin(&PinCfg);
        PinCfg.Pinnum = 29;
        PINSEL_ConfigPin(&PinCfg);
      #elif defined(LPC_PINCFG_UART3_P0_25)
        PinCfg.Funcnum = 3;
        PinCfg.Portnum = 0;
        PinCfg.Pinnum = 25;
        PINSEL_ConfigPin(&PinCfg);
        PinCfg.Pinnum = 26;
        PINSEL_ConfigPin(&PinCfg);
      #else
        PinCfg.Funcnum = 3;
        PinCfg.Portnum = 4;
        PinCfg.Pinnum = 28;
        PINSEL_ConfigPin(&PinCfg);
        PinCfg.Pinnum = 29;
        PINSEL_ConfigPin(&PinCfg);
      #endif

gerritv, 2021-02-22:
I came across this thread to find out why I couldn't get my EPSP01S working with the SKR 1.4
I can confirm that setting #define SERIAL_PORT_2 3 in Configuration.h of Marlin 2.0.x for the SKR V1.4 Turbo works great. No other changes required in other parts of the ecosystem.

Drzet, 2021-02-22:
Serial port speed needs to be the same in Marlin and ESP firmware. Learned it hard way.

rudranshsinghmahra, 2024-10-07:
My Configuration is as follows, **Board** : SKR 1.4 and **Wifi Module** : ESP01. The ESP01 works fine, as long it is plugged into a USB Programmer (not in Programm-Mode). I can access the ESP via IP from any computer on my network.
But when I plug the ESP01 into my Board's wifi port, the ESP will not be found in my Router and cannot get reached/accessed by IP. Blue light (is guess wifi light) blinks when plugged in the Usb Programmer but does not blink when connected directly to the board wifi socket. I would really appreciate if someone could help me with the solution

setayeshk, 2025-01-05:
> My Configuration is as follows, **Board** : SKR 1.4 and **Wifi Module** : ESP01. The ESP01 works fine, as long it is plugged into a USB Programmer (not in Programm-Mode). I can access the ESP via IP [...]

Did you find a solution?
I’m facing the same issue and nothing seems to be working. I would really appreciate it if you could share the solution if you have found one.

er-ebrahimi, 2025-01-05:
> My Configuration is as follows, **Board** : SKR 1.4 and **Wifi Module** : ESP01. The ESP01 works fine, as long it is plugged into a USB Programmer (not in Programm-Mode). I can access the ESP via IP [...]

I resolved this issue by connecting the CH_PD pin on the ESP8266 to a 3.3V. The CH_PD pin must be pulled high to enable the chip. However, as indicated in the SKR 1.4 datasheet, the CH_PD pin is marked as NC (not connected). It's unclear why the design was implemented this way or if there’s an alternative solution to address this issue without physically connecting the pins. If anyone discovers a workaround, I would greatly appreciate it.

rudranshsinghmahra, 2025-01-05:
> Did you find a solution? I’m facing the same issue and nothing seems to be working. I would really appreciate it if you could share the solution if you have found one.

Yes I have fixed it I just connected a 12K ohm resistor between VCC and Ch_PD pin
[image]