bigtreetech/BIGTREETECH-SKR-V1.3 #222: Pinout // all numbers of the Pins from the Board

goldjunge91, 2020-02-04:
im looking for the pin Numbers for the Skr 1.3 and 1.4 the Pin numbers to define the Aux-1 Pins

I would like to define the pins marked on the picture
//an example
`
#ifndef TEMP_0_PIN
  #define TEMP_0_PIN       P0_24_A1   // A1 (T1) - (68) - TEMP_0_PIN
#endif
#ifndef TEMP_1_PIN
  #define TEMP_1_PIN       P0_25_A2   // A2 (T2) - (69) - TEMP_1_PIN
#endif
#ifndef TEMP_BED_PIN
  #define TEMP_BED_PIN     P0_23_A0   // A0 (T0) - (67) - TEMP_BED_PIN
#endif
#if HOTENDS == 1 && TEMP_SENSOR_PROBE
  #define TEMP_PROBE_PIN   P0_25_A2   // TEMP_1_PIN
#endif
`
[image]

Fuglys, 2020-03-17:
me 2

bdelia, 2020-04-19:
I need this as well please!

bdelia, 2020-04-19:
I found this guide to the LPC1769/68/67/66/65/64/63 microcontroller very helpful, in case others come across this thread I hope it helps:  https://www.nxp.com/docs/en/data-sheet/LPC1769_68_67_66_65_64_63.pdf?fbclid=IwAR3QKfZhRvsdNU1BDoAlD6861QcTgNXTYBp2D8TwN1QkZDdlmCYKbTKfDlY

goldjunge91, 2020-04-19:
Really nice that you found this have u test the pins?

bdelia, 2020-04-21:
> Really nice that you found this have u test the pins?

I've confirmed they are correct.

goldjunge91, 2020-05-08:
yeah thanks 
i have make this 

#if TEMP_SENSOR_0 == 20
  #define TEMP_0_PIN  P0_03_A6
#endif

goldjunge91, 2020-05-09:
@bdelia hi
have you use for temp sensor the 20 or 21 in every setup what i make 3,3V and 5V by 21 i have 1 degree only and when i make 20 and 5v i have 139degree by 3.3v i have 24 same with the bed

goldjunge91, 2020-05-09:
i test this with the pins P0_03_A6 P0_02_A7