bigtreetech/BIGTREETECH-SKR-V1.3 #223: E1 Overtemp Error,  E1 not enabled in firmware, nothing plugged into E1 sensor

crm552, 2020-02-04:
My SKR 1.4 Turbo (the first one was defective) worked fine for about 3 prints, and now it won't boot.  At startup it displays an overtemp error for E1 that is persistent despite resetting the board.  Just now I tested the thermistor sockets on the board itself.  The E1 sensor socket tests as open, however the E0 sensor socket tests with just 5 ohms of resistance.  Is this a defect on the board itself?  I have tested the thermistor, replaced the thermistor, double-checked my config files.  Is this a common problem?  The defective board I got before this one showed the same error once, and then a blank screen on every other attempt.

crm552, 2020-02-06:
This is the second board I have received that has not worked.  The first one would not show up as a USB device and would not run new firmware, the seller confirmed it was a defective board.  This one came with an E1 temp error.  No reply from manufacturer regarding either issue.  Unless you are much luckier than I am, stay away from the SKR 1.4 until they figure a few more things out.

Sundancer78, 2020-02-18:
Hi, I also had the problem with E1 overtemp error after trying to plant the SKR 1.4 in my Ender 3. The temperature of the bed and nozzle jumped constantly between 22 degrees and -14 degrees. Thermistors were OK and the configuration files showed no visible errors. I reconfigured Marlin several times, all to no avail. Then I remembered that the SKR 1.4 was already working with my Geeetech A10M, so quickly plugged the thermistors from the A10M to the SKR, the error occurred again, ender thermistors plugged in again, still E1 overtemp error. Since Ender 3 and A10 are very similar, I simply played the firmware from the A10 back on the SKR and looked, no more E1 overtemp error. I compared both configurations of the firmware, but found no error. Then I copied the new configuration files of the Ender3 that Marlin supplied into a fresh directory, changed the board and had a new firmware created and flashed it onto the SKR. However, I forgot to change the LCD controller, so no picture. Curiously I switched from Marlin mode to touch mode (a TFT35 V3 is connected), the touch mode did not work all the time, that means the controller was not connected to the printer, but now it suddenly worked. Insane!
in Germany it is 7 a.m. right now, I have to go to bed slowly and will resort to Marlin again later.
Maybe my "discovery" will help someone with the same problem ... greetings .....

SkysoulPrt, 2020-03-28:
I have bought 2 boards for 2 different printers, configure in different ways, and in both, after upload directly with usb, and tweaking details , have the same problems and no longer recognized, also did put original bin back and it doesn't seem to reflash, i trying plug and unplug with new thermistor and no luck. so since my next step will sent them back, i want to try any other known firmware.bin that  could correct it. So your configuration firmware seems to get it back working again, if is not inconvenient can share with us your firmware that role back the errors @Sundancer78

Sundancer78, 2020-03-30:
Hello,

unfortunately I no longer have the SKR-1.3 in operation. But I still found
a firmware, here is the link for download

  https://yadi.sk/d/qt2sD-M9x62knA

Attached are also the configuration files from marlin from which the
firmware was created.
Please let me know if your problem could be fixed.

Greetings, André

SkysoulPrt, 2020-03-30:
Thank you al tough is not the same firmware i put the bin on  skr1.4 and managed to remove the flashing errors, i could rebuilt the newconfig

Many thanks, 
Pedro

mdellen, 2020-05-29:
When compiling the exact same source code today I get the E1 max temp error. When I load the .bin from a few months ago (same source) everything is fine. So it seems something has changed in the libraries and not Marlin itself.

I need to change something in the config but am not able to compile new firmware without getting the error. Tried BTT 2.0.1 branch, 2.0.1, 2.0.4.4 and 2.0.5.2