bigtreetech/BIGTREETECH-SKR-V1.3 #228: Fan Pin 1 Error.

Talha909, 2020-02-08:
In  Configuration_adv.h fan pin is define as #define E0_AUTO_FAN_PIN -1 but in pins_BTT_SKR.h the fine pin is define as  #define FAN_PIN          P2_03.  How can i correct one.  But i write P2_03 instead of -1 its give me the error .

Tomthi, 2020-02-09:
I was never able to get the extruder to work on that pin, If you have one Hotend, you are supposed to set the extruder fan to Pin 2.4 (HE1) using the second heater terminals for the fan. 

I have dual extruders so Fan 2.3 pin is my part cooling fan and my hotend cooling is trigged using extra endstop pins.  I have a single fan for both hotend cooling. 

* Multiple extruders can be assigned to the same pin in which case
 * the fan will turn on when any selected extruder is above the threshold.
 */
#define E0_AUTO_FAN_PIN P1_25 // -1
#define E1_AUTO_FAN_PIN P1_25 // -1
#define E2_AUTO_FAN_PIN -1

Talha909, 2020-02-10:
[image]
[image]
How i used P2_03 as a extruder fan

Tomthi, 2020-02-11:
Wait. If you're using P2.3 for the HE cooling, what are you using for Part cooling?

Talha909, 2020-02-11:
> Wait. If you're using P2.3 for the HE cooling, what are you using for Part cooling?

Part cooling is on the other port.  this one runs all the time. I want to configure P2.3 for the HE cooloing

Talha909, 2020-02-11:
[image]
[image]
I got the fallowing error when i do this. How i correct this one

Tomthi, 2020-02-12:
I don't think that's a supported configuration. 
If you want to have our HE cooling set to the auto pin, you have to use HE1 pin (p2.4) 

Use the CNC FAN pin for your part cooling.  (P2.3)