bigtreetech/BIGTREETECH-SKR-V1.3 #229: Dual-Z steppers - SKR 1.4 Turbo

scottypres, 2020-02-08:
Just got the skr 1.4 turbo and have been pulling my teeth trying to configure it to work with my Ender 5 plus. The skr 1.4 has Z0 and Z1 in addition to E0 and E1. Z0 and Z1 appear to run off the same stepper driver, but I can't find any information on how to enable Z1. I have enabled #define NUM_Z_STEPPER_DRIVERS 2 and am running all my drivers in UART with jumpers set properly. How do I enable Z1, only one stepper is moving. Does this require 2 z-endstops?

Talha909, 2020-02-08:
> Just got the skr 1.4 turbo and have been pulling my teeth trying to configure it to work with my Ender 5 plus. The skr 1.4 has Z0 and Z1 in addition to E0 and E1. Z0 and Z1 appear to run off the [...]

see this video 
https://www.youtube.com/watch?v=sZuyGD4ciwk

PowerEditon, 2020-02-24:
Hi, you are on the right way. The second 'Z2' Motor must plug to the E1  Port and UART make after your changes all needed things. You need only the jumper on MS3. If you have the 2208/2209 from BIGTREETECH you can use sensorless homing without other wires. If you want use endstops you need to cut the drag pin. And the Z2 Endstop are at 'Y_Max' define #define USE_YMAX_PLUG.
[image]

Docomar750, 2020-03-02:
Hi 
I have TMC5160 with this board and motors plugged into Z0 and Z1 are working but I am not sure if the vref needs upping as the stepper is now driving 2 motors? Any thoughts?

PowerEditon, 2020-03-02:
> Hi
> I have TMC5160 with this board and motors plugged into Z0 and Z1 are working but I am not sure if the vref needs upping as the stepper is now driving 2 motors? Any thoughts?

Hi, yes of course! The 5160 can work with 60 V & up to 20A so you can say he has no limit. If you use no UART Mode then you need to set the vref up to the double of the vref for the first motor

Docomar750, 2020-03-02:
> > Hi
> > I have TMC5160 with this board and motors plugged into Z0 and Z1 are working but I am not sure if the vref needs upping as the stepper is now driving 2 motors? Any thoughts?
> 
> Hi, yes of course! The 5160 can work with 60 V & up to 20A so you can say he has no limit. If you use no UART Mode then you need to set the vref up to the double of the vref for the first motor.
> Better idea is the second one on E1

Hi
Thanks for that

poporeto, 2020-04-11:
> Just got the skr 1.4 turbo and have been pulling my teeth trying to configure it to work with my Ender 5 plus. The skr 1.4 has Z0 and Z1 in addition to E0 and E1. Z0 and Z1 appear to run off the [...]

I have de same problem. I use tmc2208 Standalone.

nortonlam, 2020-04-11:
I'm not sure if you go the answer to your question, but the short answer is that the NUM_Z_STEPPER_DRIVERS should be 1.  The board controls both motors as one motor, so you only need to config Z0.

shawnsm, 2020-04-29:
Ok, hopefully this clarifies a bit more from what I have gotten via "translation" and trial and error regarding dual Z motors.

Bottom line, if you are using ONE driver for TWO Z Motors then plug Z1 into Z1 and the 2nd motor into Z2.  NO DECLARATION in Marlin as it's an internal wiring split on the board, set Marlin as 1 Z and thats it...
`
Rough diagram _X_ _Y_ _Z1_ _Z2_ _E0_ _Null_
`
Set Configuration.h like:
#define X_DRIVER_TYPE  TMC2209
#define Y_DRIVER_TYPE  TMC2209
#define Z_DRIVER_TYPE  TMC2209  // This would drive BOTH Z Motors plugged in at Z1 and Z2
#define E0_DRIVER_TYPE TMC2209

My single motor dual Z Marlin Configuration_adv.h:
// For Z set the number of stepper drivers
//
#define NUM_Z_STEPPER_DRIVERS 1   // (1-4) Z options change based on how many
#if NUM_Z_STEPPER_DRIVERS > 1
  //#define Z_MULTI_ENDSTOPS
  Bla, Bla, Bla
******************************************************************************

If running TWO drivers for TWO Z motors, one for EACH Z axis then plug Z1 into Z1 and Z2 into E1.   See note below regarding endstops...
`
Rough Diagram _X_ _Y_ _Z1_ _Null_ _E0_ _Z2_
`
My DUAL DRIVER DUAL Z Motor Configuration.h:
#define X_DRIVER_TYPE  TMC2209
#define Y_DRIVER_TYPE  TMC2209
#define Z_DRIVER_TYPE  TMC2209  //  This will STILL send juice to Z2 but don't use if using 2 Drivers
#define Z2_DRIVER_TYPE TMC2209  // Instead this will fire up the E1 port for your 2nd Z
#define E0_DRIVER_TYPE TMC2209

And DUAL DRIVER Dual Motor Configuration_adv.h:
// For Z set the number of stepper drivers
//
#define NUM_Z_STEPPER_DRIVERS 2   // (1-4) Z options change based on how many
#if NUM_Z_STEPPER_DRIVERS > 1
  //#define Z_MULTI_ENDSTOPS
  Bla, Bla, Bla

I am playing with this on an Artillery Sidewinder X1, with a BL Touch.  Instead of messing with regular endstops I chose to use Z-Min as the BLTouch Probe pin for my Z Probe and End Stop.  You can do this by pretending to just have one Z motor or go on to configure the G34 lines in configuration.h, As long as your HW is the same and VRefs are set the same you can just use this and BL-Touch Home off the center of the bed just fine.

You Could define 2 endstops, however I'd STILL recommend using the BL Touch in this case and using Marlin 2.0.5.3 you can play with this code if you remove the belt (on Sidewinders/Genius's) that connect them to set individual Z-offsets.  This is not needed but a cool feature.

When I have more time I'll post my full Marlin configs and pictures here and on Thingyverse.  User name ShawnM, give me a bit as I said I'm getting the printer up and going as I typed this so it may be a few before I'm able to post.  Will also have NEO Pixel info and using "extra" pins for fan control as these are things that bugged me...

Best Regards,

ShawnsM

JRosello, 2020-04-29:
@shawnsm Thanks... I was plannig to apply this config without BLTouch in my CR10s when the new SKR 1.4 arrives... I assume that it will be no difference with this setup between your Sidewinder and the CR10s, right?

Thanks again

shawnsm, 2020-05-04:
@JRosello it should be damn near the same, other than E-Step values I think.  I've got mine printing great finally, now some annoying little issues with the skr 1.4 not jiving with the tft 3.5 v3 with SD card slots and what not.

It's still working via Octoprint and the TFT works however it only sees its internal sdisk atm.  I'm actually about to compile bugfix and see what I can do about it then do it all over again for my Ender.

Take Care,
Shawn

JRosello, 2020-05-04:
@shawnsm Thanks!! I'll as SKR arrives... damn a month and a half waiting.

sjetton, 2020-05-06:
@shawnsm any chance you can share your configs? I'm doing the same mod on my SWX1 and having some issues.

C4ptAwesome, 2020-08-02:
I'd like to jump in here for a second. I'm using dual Z with the TMC2130 on all axis. Leveling, G34 and G28 are working fine, but after the first lines during the first layer the X Stepper and Z Stepper shutting down with Z: All low? TMC lib is 0.7.1

oz1sms, 2021-02-24:
I have a problem, the 2nd Z-axis motor on E1 is stepping twice as much as the first on Z0
all drivers is tcm2209 in uart mode
the stepper motor is same type. 
can I set the steps for Z1 different from Z0

seiler1505, 2021-03-02:
> I have a problem, the 2nd Z-axis motor on E1 is stepping twice as much as the first on Z0
> all drivers is tcm2209 in uart mode
> the stepper motor is same type.
> can I set the steps for Z1 different from Z0

maybe pitch settings (spindle) or the DEFAULT_AXIS_STEPS_PER_UNIT

kimkloster, 2021-04-30:
I can see I did some error  on Z1   but correcting them to this  did nothing better.
//  Z2= E1 : KK  30/4/2021
#define Z1_STEP_PIN                         P1_1
#define Z1_DIR_PIN                          P1_14
#define Z1_ENABLE_PIN                       P1_16
#ifndef Z1_CS_PIN
  #define Z1_CS_PIN                         P1_01
#endif

kimkloster, 2021-04-30:
Sorry for messing up,  maybe it can help other.  What seems to work is this....
//
// Steppers
//
#define X_STEP_PIN                         P2_02
#define X_DIR_PIN                          P2_06
#define X_ENABLE_PIN                       P2_01
#ifndef X_CS_PIN
  #define X_CS_PIN                         P1_10
#endif

#define Y_STEP_PIN                         P0_19
#define Y_DIR_PIN                          P0_20
#define Y_ENABLE_PIN                       P2_08
#ifndef Y_CS_PIN
  #define Y_CS_PIN                         P1_09
#endif

#define Z_STEP_PIN                         P0_22
#define Z_DIR_PIN                          P2_11
#define Z_ENABLE_PIN                       P0_21
#ifndef Z_CS_PIN
  #define Z_CS_PIN                         P1_08
#endif

//  Z2= E1 : KK  30/4/2021
#define Z2_STEP_PIN                         P1_15
#define Z2_DIR_PIN                          P1_14
#define Z2_ENABLE_PIN                       P1_16
#ifndef Z2_CS_PIN
  #define Z2_CS_PIN                         P1_01
#endif

#define E0_STEP_PIN                        P2_13
#define E0_DIR_PIN                         P0_11
#define E0_ENABLE_PIN                      P2_12
#ifndef E0_CS_PIN
  #define E0_CS_PIN                        P1_04
#endif

// orginal 'removed'
//#define E1_STEP_PIN                        P1_15
//#define E1_DIR_PIN                         P1_14
//#define E1_ENABLE_PIN                      P1_16
//#ifndef E1_CS_PIN
//  #define E1_CS_PIN                        P1_01
//#endif

NateCroix, 2021-05-24:
> Z2_STEP_PIN

Looking in my pins.h, it looks like there's already some conditional logic to use the next available extruder spot for Z2