bigtreetech/BIGTREETECH-SKR-V1.3 #230: USB and TFT interface not working

sergvp, 2020-02-09:
This is common problem for SKR 1.3 and SKR 1.4 
I spend a lot of time before I found that in case of shorted temperature sensor 100k (in my case was 45 ohm) display lost connection with printer and USB connection to computer became to be not recognized. I tried both of boards and found same problem.   Sorry for the time spent

PapiOrso, 2020-02-10:
I solved the problem by reloading the original firmware provided in the micro SD. I used a usb card reader to rename the original file from .CUR to .BIN then I inserted the SD in the board.

wabushi, 2020-02-20:
I have the same problem (using SKR 1.4 Turbo), TFT and USB access don't work. when I try to reflash using the sd, .CUR and .BIN won't change when inserted to the board.

Currently it feels like I fried my board, hopefully someone can prove me wrong,

soperskaft, 2020-11-07:
Does anyone have the original file for the 1.4 ?