bigtreetech/BIGTREETECH-SKR-V1.3 #231: Bed thermal run away triggered 10deg from target.

WookieWrath, 2020-02-09:
I am upgrading my CR-10 with the SKR 1.3, everything seems to be working ok but when I try to heat up the bed the thermal run away keeps triggering 10deg from the target.

For example if the target is 60c the thermal run away will trigger when it gets to 50c.
But if I set the target at 70c it will trigger at 60c and this is with any target temp.

This happens if I sent the command from the slicer or set it via the board.
what could cause this?

Tomthi, 2020-02-09:
I had similar issues with my 24v bed. Usually it's your PID tuning. I eventually had to abandon PID and use bang bang for bed heating because when I used PID it would take the bed 20 min to get to temp and it was really slow.  It would crawl when it got to with in 10 degrees of target. 

If you're hitting the thermal run away, it's probably an issue with your PID settings.  You can also increase your watch window or decrease your hysteresis value. Set your max bed power to 255 and re-run PID tuning.

WookieWrath, 2020-02-10:
I think that did it. Disabled pid for the bed and it is now holding at the Target temps.

closed 2020-02-10 by WookieWrath