bigtreetech/BIGTREETECH-SKR-V1.3 #232: SKR1.4 and BLTouch Probe

andrebu89, 2020-02-10:
Hi Guys,

i received my new SKR1.4 Board and tried to swap it agains my defective SKR1.3. I Tried :D.

Everything is working now, but i wanted to mention an Issue i ran into with the BLTouch and the new PIN Header for it.

It seems that the Configuration on this Board for the BLTouch is much different than the other boards (where the probe is plugged in in the Z-Min Header).

I had 2 different Approaches to get this thing to work without Luck:

First: Use the BLtouch Header and configure it like on the SKR1.3 Board with 
**_#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN_** enabled

My Problem: 
The Bltouch Starting as expected and deploys its pin, but quickly stows after this and the Z-Axis isnt moving towards the Nozzle. 
M119 gives a Triggered State for Z-Min

Second: Tried to declare the P0_10 pin in 
**_#define Z_MIN_PROBE_PIN P0_10_**

Problem: The Bltouch wont get an Signal to deploy its Pin but the Z-Axis is still moving to Nozzle cause the Z-Probe-endstop is showing "open"

So I Think there is some Software Issue in Combination with the dedicated BLTouch Z-Min Header.

My Workaround Solution for now, which is working well is:

Set everything normal like in the Manuals listed for BLTouch. BUT change one Line in the pins_BTT_SKR_V1_4.h file.

[image]

just change the Z_STOP_PIN to P0_10 (the bltouch pin header)

This is because in the File pins.h the Z_STOP_PIN is declared as Z_MIN_PIN

[image]

Thats why in my First attempt the Z-Min Endstop was triggered and not Open when the BLTouch tried to Level. Cause the Z-Min endstop was unassigned.

I Hope Someone will step over this Issue. Maybe there is someone who's got a nicer Workaround. At Least my BLTouch is working again and my Printer is ready to Print :)

stantond, 2020-02-11:
Search the issues in this repo, there's already an issue here with a solution

lijinlin19970813, 2020-02-18:
> Hi Guys,
> 
> i received my new SKR1.4 Board and tried to swap it agains my defective SKR1.3. I Tried :D.
> 
> Everything is working now, but i wanted to mention an Issue i ran into with the BLTouch and the new PIN Header for it.
> 
> It seems that the Configuration on this Board for the BLTouch is much different than the other boards (where the probe is plugged in in the Z-Min Header).
> 
> I had 2 different Approaches to get this thing to work without Luck:
> 
> First: Use the BLtouch Header and configure it like on the SKR1.3 Board with
> **_#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN_** enabled
> 
> My Problem:
> The Bltouch Starting as expected and deploys its pin, but quickly stows after this and the Z-Axis isnt moving towards the Nozzle.
> M119 gives a Triggered State for Z-Min
> 
> Second: Tried to declare the P0_10 pin in
> **_#define Z_MIN_PROBE_PIN P0_10_**
> 
> Problem: The Bltouch wont get an Signal to deploy its Pin but the Z-Axis is still moving to Nozzle cause the Z-Probe-endstop is showing "open"
> 
> So I Think there is some Software Issue in Combination with the dedicated BLTouch Z-Min Header.
> 
> My Workaround Solution for now, which is working well is:
> 
> Set everything normal like in the Manuals listed for BLTouch. BUT change one Line in the pins_BTT_SKR_V1_4.h file.
> 
> [image]
> 
> just change the Z_STOP_PIN to P0_10 (the bltouch pin header)
> 
> This is because in the File pins.h the Z_MIN_PIN is declared as Z_STOP_PIN
> 
> [image]
> 
> Thats why in my First attempt the Z-Min Endstop was triggered and not Open when the BLTouch tried to Level. Cause the Z-Min endstop was unassigned.
> 
> I Hope Someone will step over this Issue. Maybe there is someone who's got a nicer Workaround. At Least my BLTouch is working again and my Printer is ready to Print :)

I think you should disable Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN in the configuration.h file (the default is on), and then use the default bltouch pin in the pin file.

thisismike97, 2020-02-24:
@superperf If you are still having issues with your bltouch take a look at my config, i got it all working recently. (SKR 1.4 turbo and bltouch smart v3.1)

**Remember to close this issue if your problem has been resolved.**

closed 2020-03-12 by superperf

cronos45, 2020-03-12:
@thisismike97 Where may I find said config sir?

thisismike97, 2020-03-12:
@cronos45 My marlin fork on github has my configs.