bigtreetech/BIGTREETECH-SKR-V1.3 #233: skr 1.4 marlin, cant home the print-head

hotridejoe, 2020-02-11:
I am retrofitting a Creality Cr-X with an SKR 1.4 board. I have triied just about everything here guys. here is the problem, I cant home the the print head. i have an skr 1.4 and im running marlin 2.1 and i have standard switches for my end-stops. when i open pronterface and run the m119 command i cant get the switches to trigger I have even unplugged the switches from the skr 1.4 and still it will not respond i have tried placing a jumper between the ground and signal pins to see if it was a bad switch and still nothing. i skimmed the marlin configuration.h file for any clues to see if maybe the endstop was tuned off but it says that it is active. bottom line the skr 1.4 dosent seem to want to register the switch being pressed.  I am at a loss please help!!!, should i buy a new board?

hotridejoe, 2020-02-11:
after removing my tmc2209 stepper drivers the endstops now respond normally, does anyone know why? to my knowledge these stepper drivers are compatible with the skr 1.4. is there a way to fix this in the marlin software?

zLows, 2020-02-11:
When using the TMC2209, if you don’t use the stallguard function, you need to cut off the DIAG pin on the driver so that the mechanical switch can work normally. [Bottom of page 7](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.4/Hardware/BTT%20SKR%20V1.4%20Instruction%20Manual.pdf)

Pin can also be bended or so. Anyway, for now there is no software way to fix it. :(

lijinlin19970813, 2020-02-18:
> after removing my tmc2209 stepper drivers the endstops now respond normally, does anyone know why? to my knowledge these stepper drivers are compatible with the skr 1.4. is there a way to fix this in the marlin software?
The solution to the problem situation is as zLows said.