bigtreetech/BIGTREETECH-SKR-V1.3 #234: Neopixel on SKR 1.4 Turbo

chsc, 2020-02-11:
I'm having trouble adding some Neopixel LEDs to my printer.
Which NEOPIXEL_PIN and NEOPIXEL_TYPE should I use? It's a SKR 1.4 Turbo.
Thank you!

TAOGde, 2020-02-11:
This Board has an NEOPixel Pin Onboard.

mit freundlichem Gruß

Torsten Goltz

zLows, 2020-02-11:
NEOPIXEL_PIN    P1_21
NEOPIXEL_TYPE depends on LEDs used, more info can be found in datasheet.

chsc, 2020-02-11:
Thank you! I found **P1_24** to be working. I realized that the pins are on the board pdfs. Stupid me. **P1_20** is a pin on the EXT1 connector. 
IMHO it should be preconfigured in the _Configuration.h_ file for the SKR 1.4 board.

zLows, 2020-02-11:
Yes, of course 1_24, 1_21 is for FYSETC MINI 12864 TFT LED, sorry.
BTW, all pins preconfigured [here](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.4/Firmware/Marlin-2.0.x-SKR-V1.4-Turbo/Marlin/src/pins/lpc1769/pins_BTT_SKR_V1_4_Turbo.h)

chsc, 2020-02-11:
Thank you! Closed.

closed 2020-02-11 by chsc

Michil783, 2020-07-26:
I also have an issue with neopixel LED strip. I configured port P1_24 to be used and I also have the DCDC board plugged in. The jumper is set to use the power from the DCDC board and I do not have any output on the LEDs. The LEDs are of Type WS2812B. Configuration.h and Configuration_adv.h are  changed accordingly

Configuration.h:
// Support for Adafruit Neopixel LED driver
#define NEOPIXEL_LED
#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_TYPE   NEO_GRB // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
  #define NEOPIXEL_PIN    P1_24       // LED driving pin
  //#define NEOPIXEL2_TYPE NEOPIXEL_TYPE
  //#define NEOPIXEL2_PIN    5
  #define NEOPIXEL_PIXELS 3       // Number of LEDs in the strip, larger of 2 strips if 2 neopixel strips are used
  //#define NEOPIXEL_IS_SEQUENTIAL   // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
  #define NEOPIXEL_BRIGHTNESS 127  // Initial brightness (0-255)
  #define NEOPIXEL_STARTUP_TEST  // Cycle through colors at startup

  // Use a single Neopixel LED for static (background) lighting
  //#define NEOPIXEL_BKGD_LED_INDEX  0               // Index of the LED to use
  //#define NEOPIXEL_BKGD_COLOR { 255, 255, 255, 0 } // R, G, B, W
#endif

Configuration_adv.h

/**
 * M355 Case Light on-off / brightness
 */
#define CASE_LIGHT_ENABLE
#if ENABLED(CASE_LIGHT_ENABLE)
  #define CASE_LIGHT_PIN P1_24                  // Override the default pin if needed
  #define INVERT_CASE_LIGHT false             // Set true if Case Light is ON when pin is LOW
  #define CASE_LIGHT_DEFAULT_ON true          // Set default power-up state on
  #define CASE_LIGHT_DEFAULT_BRIGHTNESS 105   // Set default power-up brightness (0-255, requires PWM pin)
  //#define CASE_LIGHT_MAX_PWM 128            // Limit pwm
  #define CASE_LIGHT_MENU                   // Add Case Light options to the LCD menu
  //#define CASE_LIGHT_NO_BRIGHTNESS          // Disable brightness control. Enable for non-PWM lighting.
  #define CASE_LIGHT_USE_NEOPIXEL           // Use Neopixel LED as case light, requires NEOPIXEL_LED.
  #if ENABLED(CASE_LIGHT_USE_NEOPIXEL)
    #define CASE_LIGHT_NEOPIXEL_COLOR { 255, 255, 255, 255 } // { Red, Green, Blue, White }
  #endif
#endif

No output on the LED strip. Currently no steppers, hotends or anything else beside 2 fans and the LED strip are connected.
Does this causes the not functioning LEDs? I want to check as much as possible before changing the motherboard of the printer.

Any hints are appriciated.

BR Michael

TAOGde, 2020-07-26:
> I also have an issue with neopixel LED strip. I configured port P1_24 to be used and I also have the DCDC board plugged in. The jumper is set to use the power from the DCDC board and I do not have any output on the LEDs. The LEDs are of Type WS2812B. Configuration.h and Configuration_adv.h are changed accordingly
> 
> Configuration.h:
> // Support for Adafruit Neopixel LED driver
> #define NEOPIXEL_LED
> #if ENABLED(NEOPIXEL_LED)
> #define NEOPIXEL_TYPE NEO_GRB // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
> #define NEOPIXEL_PIN P1_24 // LED driving pin
> //#define NEOPIXEL2_TYPE NEOPIXEL_TYPE
> //#define NEOPIXEL2_PIN 5
> #define NEOPIXEL_PIXELS 3 // Number of LEDs in the strip, larger of 2 strips if 2 neopixel strips are used
> //#define NEOPIXEL_IS_SEQUENTIAL // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
> #define NEOPIXEL_BRIGHTNESS 127 // Initial brightness (0-255)
> #define NEOPIXEL_STARTUP_TEST // Cycle through colors at startup
> 
> // Use a single Neopixel LED for static (background) lighting
> //#define NEOPIXEL_BKGD_LED_INDEX 0 // Index of the LED to use
> //#define NEOPIXEL_BKGD_COLOR { 255, 255, 255, 0 } // R, G, B, W
> #endif
> 
> Configuration_adv.h
> 
> /**
> 
> * M355 Case Light on-off / brightness
>   */
> 
> #define CASE_LIGHT_ENABLE
> #if ENABLED(CASE_LIGHT_ENABLE)
> #define CASE_LIGHT_PIN P1_24 // Override the default pin if needed
> #define INVERT_CASE_LIGHT false // Set true if Case Light is ON when pin is LOW
> #define CASE_LIGHT_DEFAULT_ON true // Set default power-up state on
> #define CASE_LIGHT_DEFAULT_BRIGHTNESS 105 // Set default power-up brightness (0-255, requires PWM pin)
> //#define CASE_LIGHT_MAX_PWM 128 // Limit pwm
> #define CASE_LIGHT_MENU // Add Case Light options to the LCD menu
> //#define CASE_LIGHT_NO_BRIGHTNESS // Disable brightness control. Enable for non-PWM lighting.
> #define CASE_LIGHT_USE_NEOPIXEL // Use Neopixel LED as case light, requires NEOPIXEL_LED.
> #if ENABLED(CASE_LIGHT_USE_NEOPIXEL)
> #define CASE_LIGHT_NEOPIXEL_COLOR { 255, 255, 255, 255 } // { Red, Green, Blue, White }
> #endif
> #endif
> 
> No output on the LED strip. Currently no steppers, hotends or anything else beside 2 fans and the LED strip are connected.
> Does this causes the not functioning LEDs? I want to check as much as possible before changing the motherboard of the printer.
> 
> I also tried other output pins like the servo pin (P2_00). No success.
> 
> Any hints are appriciated.
> 
> BR Michael

Case_light AND NEO_Pixel on the same PIN ???? Thats are two different things. Change the Case_light to an free PIN.

Michil783, 2020-07-26:
Hi,
thanks for the tip but it does not change anything. tried to put case light to P1_24 and Neopixel pin to P2_00 and vize vorsa. No change. No lights.

TAOGde, 2020-07-26:
> Hi,
> thanks for the tip but it does not change anything. tried to put case light to P1_24 and Neopixel pin to P2_00 and vize vorsa. No change. No lights.

Neopixel do  the startup Test? When not is the power and ore booth data signal Wrong

Michil783, 2020-07-27:
nothing, there is no LED reaction. Even nothing during startup test. regardless which output pin I'm using. LED strip itself is working (tested with another HW). I assume some configuration error. Is there any other config I have to switch on?

NinjaJake86, 2020-07-27:
If you are using the Turbo board then try disabling the case light and setting the neopixel to use P1_24 (If you are using the built in neopixel connection that is)

Michil783, 2020-07-27:
I got it now running on Pin P2_00 (Servo connector) after deleting the Adafruit_Neopixel compile files and forcing VCode to recompile it. I assume the Pin P1_24 is defect. I will exchange the board and see if this is the case. Case light is currently disabled.

javicarrera, 2021-01-19:
I spent 2 days looking for this. My neopixel pin P1_24 doesn't work either, I put it on another endstop free pin and finally works. Did you try with another board ? Does it happen to everyone?

joefarmer, 2021-03-13:
> I spent 2 days looking for this. My neopixel pin P1_24 doesn't work either, I put it on another endstop free pin and finally works. Did you try with another board ? Does it happen to everyone?

I realize this is a closed issue, but I had the same issue on mine. The P1_24 for NEOPIXEL didn't work and I had to move to PWRDET P1_00.

Tuckpolo, 2022-03-14:
Hi stumbled on this chain dealing with the same problem. I was getting ~.8VDC on the PWR pin. Turns out, my SKR 1.4 boards did not have jumpers on the red pins as shown below. I have a separate external 5VDC power supply. By installing the jumper, my VDC pin went to 5VDC and they turned on.
[image]

https://user-images.githubusercontent.com/31577989/158256852-c1bca139-685c-4655-beb0-d2020f014164.png