bigtreetech/BIGTREETECH-SKR-V1.3 #235: SKR v1.4 Turbo PNP NO probe

Alyxxik, 2020-02-11:
HI,  i have 11-30V capacitive probe i been using on Ramps 1.4 till i buy SKR v1.4 Turbo for my Corexy printer. But with this board it doesnt work. 

 Its PNP NO probe connected to 12v PSU and signal wire connected to Z endstop pin ( i used 10k and 12k resistors to decrease voltage to 4,8V, that work fine with Ramps). But no matter what, M119 report "Open"  even if the proble is triggered. 
 
 I measure voltage on ground and Signal wire disconnected from the board, there is nothing, when the probe is trigger i have 4,78V on signal wire. But after i pluged it in the board (1.27 Z endstop pin),  i get readings  0.247V when open and 0.260V when trigger (endstop interupt change nothing).
 Is there some conflict with some internal pullup/down resistors?  (mechanical endstop work fine) 

 Also, i been thinking, how much voltage can signal pin take?  I mean can i plug in the probe directly to the board without any voltage divider whatsoever? I currently use old 500W PSU from old computer,  but i have 400W 24V on the way.

AnubisCZ, 2020-02-15:
I have the same problem. On Anet A8 it worked well on SKR no.

Knowledge from a colleague: at logic zero (closed sensor) there must be a voltage in the range of 0V to 1V and at logic one must be greater than 2.1V. Nothing in between! Anet board was enough to identify logic zero everything below 1.6V

Alyxxik, 2020-02-16:
Ano, to sedí, ačkoliv to není zrovna bežný senzor. Je to průmyslová sonda, dražší než Anet A8 na kterou jsem si ji chtěl původně nasadit. :D 
 
 Bohužel SKR V1.4 to vůbec nezajímá, a absolutně jí ignoruje... Napadlo mě použít transistor jako spínač, který by fungoval mezi 5V a Signálem přímo na desce a sonda by fungovala jako trigger. Ale V1.4 teď bohužel už nefunguje vůbec z neznámého důvodu (přišel jsem po 5 hodinovém tisku, upravil Firmware a od té doby deska nereaguje na příkazy k pohybu. Včera se mi podařilo jí rozbýhat, ale jen Z, které vydává zvuky jako by se měl motor rozpadnou a XY neraguje), Jako rezervu jsem používal RAMPS 1.4, to ale po vyhození jističe u mě doma chcíplo, takže teď mám jen základku pro Anet A8 a SKR MINI E3 V1.2 které mám na to co původně bylo Anet A8. 
   Jaká ironie, že původní Anet A8 v1.5 deska je jediná věc co opravdu funguje. I ta MINI E3 mi šíleně pije krev...

AnubisCZ, 2020-02-20:
On the forum I got a tip to connect an optocoupler.

https://3dprint.wiki/_detail/reprap/anet/a8/improvement/15895108_1220874871337969_6770101747027532263_n.jpg

AnubisCZ, 2020-02-25:
Friends have designed this connection and everything works perfectly. I used for SKR 1.4 Turbo and inductive sensor SICK IM08-04NPS-ZTK.

[image]

Kraken3d, 2020-02-26:
While making the old inductive probe work on my new set up I had to remap the pin for the probe. Got the information from Chris Warkocki off youtube. Might not be ideal but it worked. 

[image]

phazei, 2020-04-23:
So, I needed to get an inductive probe set up as well.

I'm using TMC2209 with auto-homing for xy.

The servo plug matches my probe pin plug, and since I'm not using a BLTouch, I'm otherwise never going to use that port.  So I changed the pin order in the plug to match the GND/Pwr/2.0 of the servo plug.  Also needed to use a new setting only available in the last month on the marlin bugfix 2.x branch.

#define USE_PROBE_FOR_Z_HOMING
#define Z_MIN_PROBE_PIN P2_00

Working great for me now.  Should be documented and made default.  Just make sure to switch the gnd/pwr on the original pin to match the servo port.

JasonTurnbull25, 2020-07-31:
Anyone still dealing with this? I have LJ8A3-2-Z/BY-5V. This is a 5v probe but it IS PNP/NO. I haven't seen any definitive answers for getting it to work with an SKR 1.4 Turbo. Anyone out there have any solid info?

zozo6015, 2020-08-27:
I have a 5V inductive probe. No matter where I plug it in I always get TRIGGERED no matter if the probe is TRIGGERED or NOT. I am running on skr 1.4 turbo. Any ways I can make this work?