bigtreetech/BIGTREETECH-SKR-V1.3 #236: Question about fuses

straccio, 2020-02-12:
Good morning,
i have recently bought a 1.4 version of this board.
There are 2 fuses, 20A and 10A, i need to know if i need to change these fuses if i power the board with 24v?

Best regards.

Tomthi, 2020-02-13:
I run mine on 24v and did not change the fuse.

TobiasFechner, 2020-03-04:
There is no need to change the fuses if you run the board at 24V. However, take care not not exceed the power rating (see manual) and you should be fine.

Peemouse, 2020-06-17:
Hi,
I use this issue for asking what type of fuses are installed in v1.4 ?
I blew up the 10A one (by inverting + and - on main leads 🤦 ) and need a replacement one.
I didn't find any details about those in the documentation (unless I missed it).

@bigtreetech, could be worth to add it in the manual ?

Evanlec, 2020-08-07:
> Hi,
> I use this issue for asking what type of fuses are installed in v1.4 ?
> I blew up the 10A one (by inverting + and - on main leads  ) and need a replacement one.
> I didn't find any details about [...]

I also would like to know what type of fuse as my 10A fuse is blown on my SKR 1.4 Turbo board. Thank you.

Peemouse, 2020-08-07:
Hi @Evanlec,
I figured out that the fuses look like those ones :
https://www.mouser.fr/ProductDetail/Littelfuse/0297010WXNV?qs=bpnLQM2ZShyfiVIOvdY7nQ==

But haven't tested yet.

Evanlec, 2020-08-13:
I was able to find the correct fuses on Amazon:

https://www.amazon.com/gp/product/B01DYQCXRO/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1

senolyaya, 2021-10-30:
> Amazon'da doğru sigortaları bulabildim:
> 
> https://www.amazon.com/gp/product/B01DYQCXRO/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1

fuse is true but power security layer is shotcuted on the board. Check S1A and S1G diode with multimeter shotcut check. replace S1A  (M1 or SS14) schottky diode and 10A fuse