bigtreetech/BIGTREETECH-SKR-V1.3 #237: Can bigtreetech skr v1.3 mainboard run with 11.8v ?

Lasithalks, 2020-02-12:
I bought a 12v 15A powersupply and skr v1.3 board. But in my power Supply its output voltage is only 11.8v . Can i run my skr v1.3 mainboard from this power

Lasithalks, 2020-02-12:
[image]

chsc, 2020-02-13:
Should work. The internal step down converter on the board is stepping it down to constant 5V or 3.3V depending on the electronics/microprocessor. Unter load it might drop even more so your bed and extruder heating will take a little bit longer. The electronics itself should be unaffected by this. On some power supplies threre is a potentiometer to adjust the voltage. But this seems to be an ATX power supply, right? Be shure you use the right polarity.
Personally I don't use 12V anymore and switched to a MeanWell 24V/350W power supply.

thisismike97, 2020-02-24:
@Lasithalks 11.8 volt is within spec and should work just fine, however if you have a heated bed on your printer 15Amps might be a little on the short side..

**Remember to close this issue if your problem has been resolved.**