bigtreetech/BIGTREETECH-SKR-V1.3 #239: SKR1.3 not Connecting

BaggySW, 2020-02-15:
I bought a SKR1.3 with 2208 drivers for my CR10.

There is an error TMC connection error. Will take instructions from the display but make a terrible noise when moving on the z-axis towards the bed.

Tried to connect to use m122 command first  I tried to connect my board via usb to my Mac using pronterface with no success, then tried connecting via octopi still with no success. 

Not sure how to progress, any help greatly appreciated.

thisismike97, 2020-02-24:
@BaggySW Post your config files so we can see what's wrong.

1larity, 2020-02-29:
Have you enable the 2208  flags in marlin, by default it has Allegro A4988 drivers set.

see #define X_DRIVER_TYPE in config.h

BaggySW, 2020-03-02:
I've been playing with the Firmware and have resolved several issue, it's no longer making a terrible noise on the z-axis moving down towards the bed and I can now get octopi to connect.

Still have a TMC connection error, however everything appears to work as it should despite this.

a M122 command gives the following message
Testing X connection... Error: All LOW
Testing Y connection... Error: All LOW
Testing Z connection... Error: All LOW
Testing E connection... Error: All LOW