bigtreetech/BIGTREETECH-SKR-V1.3 #240: TFT35 V2.0 and SKR V1.3

SIG70, 2020-02-16:
Hello.
SD card error. 

Windows on the computer sees the SD card.
http://www.youtube.com/watch?v=jFBue1sybhA

rainerschulte, 2020-02-22:
One Idea I have is to check USB-Config in Marlin.

SIG70, 2020-02-22:
I tried all the options.
#define SDCARD_CONNECTION LCD
or
#define SDCARD_CONNECTION ONBOARD

Tried to include.
//#define NO_SD_HOST_DRIVE   // Disable SD Card access over USB (for security).

rainerschulte, 2020-02-23:
`#define SERIAL_PORT -1

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Serial port -1 is the USB emulated serial port, if available.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 0

/**
 * This setting determines the communication speed of the printer.
 *
 * 250000 works in most cases, but you might try a lower speed if
 * you commonly experience drop-outs during host printing.
 * You may try up to 1000000 to speed up SD file transfer.
 *
 * :[2400, 9600, 19200, 38400, 57600, 115200, 250000, 500000, 1000000]
 */
#define BAUDRATE 115200
`
`/**
 * SD CARD
 *
 * SD Card support is disabled by default. If your controller has an SD slot,
 * you must uncomment the following option or it won't work.
 *
 */
#define SDSUPPORT
`
These ar my settings. Ionly use the SD from the Display since the internal SD-Slot from the SKR1.3 is junk. It doesn't keep (or like)  SD-Cards and throw them out... Firmware updates over USB work propperly. 
Currently I work on the Connection if ESP01-S in the Display to print via Network...

SIG70, 2020-02-23:
Hello.
I did everything as you indicated.
Read Error SD flash — SKR 1.3 

> `#define SERIAL_PORT -1
> /**
> *  Select a secondary serial port on the board to use for communication with the host.
> *  This allows the connection of wireless adapters (for instance) to non-default [...]

rainerschulte, 2020-02-23:
hm... I've learned from this guy... [https://www.youtube.com/watch?v=57f-sG0nFeM](url)
I also had to start from Scratch again. Watching his Videos helped me a lot.

SIG70, 2020-02-23:
Thanks. I’ll try very carefully to repeat everything on the video.I’ll inform you about the result.

> 
>hm... I've learned from this guy...  https://www.youtube.com/watch?v=57f-sG0nFeM
>I also had to start from Scratch again. Watching his Videos helped me a lot.

SIG70, 2020-02-23:
You have working BIN files for the TFT35 V2.0 and SKR V1.3 firmware. I will try to check.

> 
>hm... I've learned from this guy...  https://www.youtube.com/watch?v=57f-sG0nFeM
>I also had to start from Scratch again. Watching his Videos helped me a lot.

SIG70, 2020-02-24:
Hello.
Here everything is briefly said about the shortcomings of this TFT35V2.0 V3.0 board.
Video in Russian.
http://youtu.be/1SpBpdtXKYU?t=5979

> 
>hm... I've learned from this guy...  https://www.youtube.com/watch?v=57f-sG0nFeM
>I also had to start from Scratch again. Watching his Videos helped me a lot.