bigtreetech/BIGTREETECH-SKR-V1.3 #243: Missing schmatics for SKR 1.4

tvixen, 2020-02-20:
Could you please upload the schmatics for the SKR 1.4 board.

Thanks in advance

thisismike97, 2020-02-24:
Do you mean this one?

https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.4/Hardware/BTT%20SKR%20V1.4PIN.pdf

paulgking, 2020-02-24:
That isn't a schematic, I think he means (and I would certainly like) a 1.4 version of this for 1.3 https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.3/hardware/SKR-V1.3-SCH.pdf

tvixen, 2020-02-24:
We already have the schematics for skr 1.3.  I would like to have it for skr 1.4.

SusiBiker, 2020-02-26:
As paulgking described above, SKR 1.4 schematics would be really useful.

AnubisCZ, 2020-02-27:
Please SKR 1.4 scheme.

jstaylor11, 2020-02-28:
I would love to have a look at this as well so I can troubleshoot some issues I am having with the SKR 1.4 Turbo board and the extruder (TMC2208 3.0) not working.

majo91, 2020-02-28:
I need it as well! @bigtreetech please upload the schematic. thanks!

asdmin, 2020-03-09:
172 open issues at writing this entry and the fact that they crammed 1.4 documentation into the 1.3 project clearly shows that BTT only uses github as a content delivery system.

In other words, you are shouting into a black hole here guys. Buy the 1.3 instead. Or any other vendor who can properly document their product.

asdmin, 2020-03-09:
dup to https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/184

DeviousPenguin, 2020-03-12:
> I would love to have a look at this as well so I can troubleshoot some issues I am having with the SKR 1.4 Turbo board and the extruder (TMC2208 3.0) not working.

Are you running Klipper firmware?

thisismike97, 2020-03-12:
@tvixen As of 7 hours ago, the schematics have been uploaded, please close this issue and goodluck with your project.

tvixen, 2020-03-12:
Thanks for upload. 

Case closed.

closed 2020-03-12 by tvixen