bigtreetech/BIGTREETECH-SKR-V1.3 #245: SKR V1.3 Stops Updating Temperature and Turns Off Hotend

CoinNester, 2020-02-21:
Hello, I am trying to figure out whether my SKR v1.3 is starting to fail. On several occasions now, the display on the printer would show lets say a setting of 209C and actual at 210C and it will stay that way. This causes the hotend to turn off completely until it is room temperature while continuing to print and making it look like everything is fine. I have changed the thermistor just in case and it has not solved the problem. What is happening? I haven't seen this issue.

CoinNester, 2020-02-21:
To provide some more information:
This is a home built Delta printer that has been working flawlessly for the past few months.
It is running Marlin 2.0.x bugfix with TMC2208. I have not been making any changes to the firmware configuration when this started happening. 

Strangely enough, I also saw that the heater for the hotend would not even turn on after the unit was turned off completely and allowed to cool down. I thought that after the reset, the controller would see that I am asking for 209C and the hot end is at 22C and it would turn on, but it would take several restarts (after Marlin popped up an error stating that the hotend is not heating) that it started to temporarily work again. I sincerely doubt that the heater itself is at fault here because of the discrepancy of the temperature being displayed that I described in the initial post. Could the mosfet be dying? I would imagine that if there was some sort of grounding issue going on with the temperature sensor, it would just start showing -14C like it always would when things are disconnected and not get stuck at the last highest temperature.

I might have to try switching from E0 to E1 and see if that solves the problem at all.

toms5556, 2020-05-10:
Same problem here - upon startup the board turns on bed and hotend when requested and turns off automatically after only a few seconds. Also thought of changing thermistor but will not now I found this post. Re-compiled and loaded firmware, no luck. Will order a spare skr to see if this is the problem. I can see that the board cancels the heating signals, it is not the mosfets. Have you solved this problem in the meantime?

CoinNester, 2020-05-11:
> Same problem here - upon startup the board turns on bed and hotend when requested and turns off automatically after only a few seconds. Also thought of changing thermistor but will not now I found [...]

It turned out to be the MOSFET controlling the hotend heater was dying or dead. I switched to the second hotend heater output and everything works fine. You'll have to adjust your firmware to make it think E1 is E0 by switching the pinouts.

toms5556, 2020-05-12:
Thank you CoinNester. I had tested the mosfet (On board) and it seems dead, the others return expected value. Now buying one is another matter altogether - with only rudimentary electronic skills it may be not so easy to determine an equivalent. This type mosfet I learnt costs more In Australia than the whole board from China... (just the delivery time). Thank you for commenting.

CoinNester, 2020-05-12:
I see. That is a tough situation. In the future you might want to consider using external mosfet board to control the heaters. I am using that on my heat bed on my other printer with an SKR v1.4. Turns out that the newer SKR can't handle the wattage and the mosfet was almost cooked. Hooking up the output to an external mosfet board solved the issue. I might do it for a hot end as well in the future to add life to the SKR. 

Here is the one I am using:

https://www.amazon.com/gp/product/B074KZF772/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1

toms5556, 2020-05-26:
Just to let you know how this ended: I tested the onboard mosfets again and they seemed ok. Thought then it must be a software controlled issue which eventually let me find that the hot end thermistor had failed which caused the mosfet not to turn on the power. Replaced hot end, all working again.

CoinNester, 2020-05-26:
Ok so the thermistor was falsely giving higher temperature numbers? Or is the software able to see when a thermistor is not giving any feedback and just shuts the heaters off?

toms5556, 2020-05-26:
I suspect the latter is the case but I am not skilled enough to find out for sure.

TomS

Ok so the thermistor was falsely giving higher temperature numbers? Or is the software able to see when a thermistor is not giving any feedback and just shuts the heaters off?