bigtreetech/BIGTREETECH-SKR-V1.3 #246: accidental short circuit, any chance of repair?

MikRoscope, 2020-02-21:
I accidentally shorted the 12/24V and GND pins on top left (near X axis motor connector) of my BTT SKR V1.3 while the board was on 24V power.. after a spark the board seems to be dead. Is there any chance a voltage regulator is burned and could be replaced? or the processor is dead? I could not find the schematics to follow the connections to that pin.

Andy-ABTec, 2020-02-22:
Hi, schematic is [HERE](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.3/hardware/SKR-V1.3-SCH.pdf) - good luck...

MikRoscope, 2020-02-22:
thanks for the link, I had a look, but not sure which part of schematics is the pin I shorted.. is that the pin connected directly to VBB in bottom left of the schematics? 
otherwise when I power the board now the red led on bottom left is lighted but very dim, and the chip marked here gets hot very fast 
https://imgur.com/a/4XWL4eF

thisismike97, 2020-02-24:
The chip you marked probably has a short around it somewhere, check around with a multimeter.

You'll probably have to replace that chip and maybe even more, fixing that chip could reveal the next problem and so on.. 

Considering the time investment and finding the correct replacement parts its probably better to just get a new board.

MikRoscope, 2020-02-24:
Thanks a lot for the tip! I will check that chip and around it. I have access to few replacement parts. 
I have already ordered a new board, until the new board arrives, I like to try to fix this one as a hobby before throwing it out! ;)

thisismike97, 2020-02-24:
@MikRoscope I also recommend testing the board using only usb power instead of your 12/24v power supply, a lot of the parts can handle the 5v usb but would die instantly on 12 volt or higher while there are still short circuits present. 

If you manage to fix it, you could always return the new board, you got nothing to loose, goodluck!

Also don't forget to close this issue after the repairs / new board.

MikRoscope, 2020-02-24:
@thisismike97 That is a good idea, I will work on USB power to check the board. 

After close inspection of the area near U4 (the chip that gets hot fast) I noticed clear burn marks on LV5, C12 and LV4 as seen the image below. LV4 and LV5 are 22microH inductors and C12 is 0.1microF capacitor based on the schematics. I am not sure about their correct smd size or rating though. If I could find them in my local electronic shop, I will replace them first. 

[image]

thisismike97, 2020-02-24:
@MikRoscope yup, i also checked the schematic just now, those burned components are part of the 3.3v rail, however maybe those components blowing up saved your SOC. though at this point i highly suspect it's beyond repair.

I'm pretty curious, let me know if you succeed.

MikRoscope, 2020-02-25:
I replaced the broken components but still U4 as well as the SOC gets very hot even on 5V usb! Unfortunately, the board is beyond repair for me!

closed 2020-02-25 by MikRoscope