bigtreetech/BIGTREETECH-SKR-V1.3 #249: SKR 1.4 Turbo issues with prints

vonryan123, 2020-02-27:
This only started with this board. Printer worked great before. Tried a few versions of marlin 2.0x even 2.04.4 which came out today.
I have tried:
Old and new slice don't matter
Baby steps disabled
Linear Advance disabled (in the firmware as well)
I have set Vrefs anywhere from stock 1.125 to .68
TMC2209's, TMC2100's and DRV8825's All working correctly no jumper issues
I'm am about ready to toss this back to amazon......Printer works but this surface issue is on everything infill, walls, skin/top surface everywhere......
[image]

1larity, 2020-02-29:
You don't say what changes you made to the firmware for your printer, for instance I fount the default Extruder steps per mm was insanely high for ender 3 which requires 95 per mm, whereas the default is 500. Overall I had to make over 50 changes to the firmware before it was appropriate for ender 3.
If you expect help or to raise a useful issue, you need to post your config files.

vonryan123, 2020-02-29:
I changed everything I needed too as far as what I need to setup for the printer. After doing more research on the matter I did find a few things.
Motor pulses in the newest firmware may still be an issue
The SKR 1.4 Turbo just now got supported and I known to be a big buggie
TMC 2209's have also shown many issues 

I will close this as most people that reply dont know what they are talking about and the ones that do don't actually help so .....lesson learn github is shit

closed 2020-02-29 by vonryan123