bigtreetech/BIGTREETECH-SKR-V1.3 #250: Skr 1.4 Turbo TMC 2209 v2 Watterott:TMC Connection Error

schuuuu, 2020-02-29:
Trying to setup tmc 2209 v2 from watterott with a skr 1.4 turbo. Followed the instructions on learn.watterott and cut off the spread pin +soldering bridge on skr + jumper is set. Used latest marlin Bugfix and also bigtreech version, but still getting TMC Connection error. Motors can be moved. 

Any suggestions to fix the issue?

TobiasFechner, 2020-02-29:
I had the same problem. Here is the solution:

You have to make a solder bridge on the underside of each TMC2209 stepper board to get the Uart signal in/ out using one of the PDN_UART pins. If you turn the board around you can see three little solder pads. If you look at the bottom side of the stepper board with the three diag pin holes oriented to to the left, just connect the middle pad with the left pad. That should do the trick.

Also the communication to the driver will only work if a power supply is connected (e.g. 12V or 24V).

Regards
Tobi

schuuuu, 2020-02-29:
Thanks Tobi. Tested it and it works fine.

Best regards 

Marvin

closed 2020-02-29 by schuuuu

kevinmorrison, 2020-04-21:
Having this issue but not sure what three pads you are talking about? I see a couple of places that have three pads and dont want to solder the wrong ones. Any better clue where these are?

schuuuu, 2020-04-22:
Hey Kevin, 

do you have TMC 2209 from Watterott and SKR 1.4 Turbo? If so just upload a picture of your stepper driver. 

Best regards 

Marvin

kevinmorrison, 2020-04-22:
Thanks Marvin, I was going to just delete this post because I discovered the problem and I am a bit embarrassed. What I figured out is the risers that the pins on the driver go into on the MB are weak and when I was pushing them onto the MB the riser was bent out just far enough that the pins slid down the inside of the riser instead of going into the holes intended for them. So of course no connection was being made. I find these types of high risers to be problematic in that they apparently can be loose from the factory. There has got to be a better way to mount these than going with these high flying connections that have weak joint to the MB. Anyway that is what happened to me and hopefully someone else having issues will read this and find the same thing was happening to them.