bigtreetech/BIGTREETECH-SKR-V1.3 #251: Skr 1.4 Turbo TMC 2209 v1.2 - Two Z steppers dont work

ROMSY, 2020-02-29:
i am replacing my board right now and trying to get my two z-steppers on one driver to work.
x and y just work fine, and also my first test with one stepper on z.
but when i put a second stepper on z its like he has not enought power (motor does only make a weird sound) what could be the issue here?

JRosello, 2020-02-29:
Interested on this issue... I'm just waiting to arrive my board to fit in my cr10s that has 2 Z steppers.

paulgking, 2020-03-01:
You might want to give a bit more detail. I'm using a 1.4 (no turbo) with TMC2209 and no problems dual z axis now. 

So what stepper driver are you using etc. ? Not that it'll guarantee help, but at the moment you are asking people to work blind to your setup other than the board.

ROMSY, 2020-03-02:
Hey, 
sorry i am very new with the 3d Printer stuff (and changing the board) The printer is an FLSUN Cube and the Stepper Motors i am using are: SL 42STH40-1504A
Unfortunatally its the info on the left motor, on the right motor there is no info at all. but after checking x and y there is also no info on on of the steppers and the second on has the same info so i guess those are all the same steppers?

So far i changed to voltage in configuration_adv from 800 to 1100 and it started to work but now those two are really noisy.
i left mostly every value on its default settings but changed  DEFAULT_AXIS_STEPS_PER_UNIT  on x and y to 160, 160 (z has still the default of 4000 maybe i should double it too?) 

since it makes a heavy noise i gonna check out later when i am back from work if its a mechanical issue with the sound.

is there more information i can /should provide?

ROMSY, 2020-03-02:
Update:
i turn on the printer and move Z... its super noisy!
i start pronterface and put in **M92 Z4000** and THEN move z, its silent as the other motors.
Question is now what could case this behaviour?

paulgking, 2020-03-03:
The times I have had very noisy Z was either mechanical where thing were too tight, or the motor current was too high. Did you increase the current pot on the motor driver when trying other things?

ROMSY, 2020-03-03:
Ok, i am sorry: i downloaded the latest version from the marlin website instead the one from github.
after reflashing this one everything works out of the box. so i just need to update the 2.0.1 to 2.0.4.4 it was all my fault ><

closed 2020-03-03 by ROMSY