bigtreetech/BIGTREETECH-SKR-V1.3 #255: USB port burns down

spammchik, 2020-03-05:
The motherboard manufacturer SKR 1.3 reports that:
> The power of the hot bed connected to the motherboard must  be less than or equal to 144W (that is, the thermal bed resistance is greater than 1Ω). If you want to use a high-power hot bed, you must connect an external hot-bed power expansion board.

I have a heated table with a resistance of **0.8 ohms**. And use a **14 volt** power supply voltage.
When I turn on the printer to heat the table, the heating goes without problems, the board is working.

However, if I connect the USB cable to the computer while the table is heating up, you can hear that something clicks in the board and even smoke comes, after which the USB Port stops working forever.

Thus, I messed up two SKR 1.3 boards. USB port no longer works on them.

Is it possible that the USB port is burned out by the fact that the heated table has a too small resistance of 0.8 Ohms?

Or do I need to look for another reason why the USB port burns out?

Al9900, 2020-03-07:
You probably didn't switch the jumper between POWER SUPPLY and USB. Why you need to connect usb while it's going?

Thorinair, 2020-03-09:
My guess is that he wants to send GCode commands over USB. Like literally everyone does.

spammchik, 2020-03-10:
> You probably didn't switch the jumper between POWER SUPPLY and USB. Why you need to connect usb while it's going?

> My guess is that he wants to send GCode commands over USB. Like literally everyone does.

Of course, the power jumper is in the INT +5V position.
At the same time, connecting the USB cable is necessary to control the printer from a computer.

spammchik, 2020-03-11:
The element D3 burned out on the board.
Why is this happening?
Four SKR 1.3 boards have already burned so badly for me.
Perhaps low-quality components for these boards are used?
Does anyone know if there are better analogues for replacing the D3 element - NUP2114UCMR6T1G ???
[image]

tarifari, 2020-11-25:
Same problem here on skr 1.4.

Burned two boards. Cant connect them anymore. But cant find D3 on my SKR 1.4

DId you solve this?

SeanRods, 2021-04-27:
Hi, I was checking if you were able to fix this. My board SKR 1.4 worked for 8 months before this issue. I am sure its the same USB issue. It works but I am unable to send commands over USB.

Ari-SSO, 2023-07-09:
Hello, I have a skr 1.3, and occasionally I noticed a smell like something burning, but this mixed with the smell of the molten filament so it didn't seem so strange to me. I thought it was the poor quality wire, so I insulated it with high temperature silicone.
I almost always use the bed at 45 degrees, the last time I heated it to 48 degrees and after a few minutes I realized that smoke was coming out of the board, I don't remember having the USB cable connected to the PC, although I always use the other end connected to the board.

Could you please tell me if this is still the case in SKR 1.4 or in octopus?

Thank you.