bigtreetech/BIGTREETECH-SKR-V1.3 #257: V1.4 Turbo Dual Z Closed-Loop Stepper

ehtnevets, 2020-03-08:
Is it possible to to connect dual Z with SB42B V.10 Closed Loop steppers to SKR 1.4 Turbo? If yes, instruction please.

Thanks

driver200, 2020-03-30:
Удалось подключить?

dimi2405, 2020-04-29:
I am interested in the same thing, i am thinking merging the whires into just ome conector could do it ? Really looking forward to a response from someone who knows what he is talking about.

ehtnevets, 2020-05-01:
I use 7417 to split the TTL signals (Step, Dir, Enable) coming out of the SKR 1.4 . The buffer conditions and amplifies to split the signals for, I think, up to 16 lines. So, 2 lines is no brainer. It works much better than using Dual motor configuration on Marlin firmware.

ehtnevets, 2020-05-01:
[20200501_040347170_iOS.heic.pdf](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/files/4562456/20200501_040347170_iOS.heic.pdf)

dimi2405, 2020-05-01:
Thank you, i will try this.

kreuzhofer, 2020-06-15:
Could you post a sketch how you wired it up? I am interested to build it but you seem to have added some additional components?

ehtnevets, 2020-06-15:
Nothing much, the capacitor is a filter that you put on Gnd and Vcc pins. The resistor is for output enable.

ehtnevets, 2020-06-15:
But I don't think anybody should use BTT closed loop system at all. It's not safe as the system causes motor to spin out of control upon re-enabling (not all the time but very often). Especially on Dual motor Z-axis. Imagine your steppers would spin full power and out of control for a few revolution. Worse if only one of the motors on your Z-axis does that.
BTT doesn't seem capable of fixing the bug as they never respond to the report.

stevehooker, 2020-07-14:
> Especially on Dual motor Z-axis. Imagine your steppers would spin full power and out of control for a few revolutions. Worse if only one of the motors on your Z-axis does that.

I have S42Bs on all 5 motors. Yes, I'm seeing this more often than not. Luckily, with the G34 - Z Steppers Auto-Alignment they can be brought back into alignment, But they are mostly **so far out** that I have to run that twice - 12 iterations, each.. I have no idea why the motors would suddenly and rapidly seem to turn for several revolutions, seemingly out of control. On the two Z axis motors they at least always spin up, so avoiding a crash into the bed. And X and Y do not crash out into max end stops. So, far it seems livable, though very undignified.

 I've only had these S42Bs running for less than a day and as yet not printed anything so I don't know of all the trouble will be worth it. Especially if/when this out of control initial spinning happens on the extruder - I've not seen this yet.

ehtnevets, 2020-07-14:
Oh yea, my dual extruder printer, because of the mechanical setup one rotates opposite the other to do exact thing. So, one extruder ejected the filament while the other pushed so hard that the slippage generated quite a bit of plastic powder.

For 3D printing, I think the smoothness of motion is most important. If losing steps is a problem than there are more problems with the printer that SB24 can solve.

stevehooker, 2020-07-14:
Arrrrg! Yes, trying to babystep my first layer... Hadn't noticed the extruder had unloaded 20cm, an entire Bowden tube... No wonder it looked like there was a clogged nozzle - no filament. Currently trying a new start up code, which extrudes 30cm, just in case. As well a balance of the Z screws, then for good measure a fresh bed levelling. Long startup as a workaround - not good.

I have Nano Zero Steppers in two resin printers and though I have THK KR26 linear actuators with a 2mm pitch and a 0.9° steppers, they're probably unnecessary, it's a nice warm feeling, knowing those 5µm layers are perfect. It's not missed steps, it's perfection - accuracy. To stop the print midway, move it up out of the resin, to look, and resume the print, exactly, exactly, perfectly exactly where you left off is splendid. 

I thought that it would be cool to have five closed loops on a new build AM8... Another few days banging my head against the wall and I think I'll drop them and go with some TMC drivers, as I have in other FDMs.

The way these S42Bs go crazy, in X, Y and 2 Z axis is terrifying. I guess I'll keep trying some workarounds, but I'm losing heart on the idea.

I saw BTT has released a new firmware update yesterday.  I guess that will be my last gasp.

ehtnevets, 2020-07-14:
If you have access to Marlin's source code (assuming you use Marlin), you can set it up so it never disables steppers. All 3D printer firmwares disable after finish and cancel job and re-enable (this is where the problem) on the first new move command by default. TMC is the way to go. So nice and quite meaning less vibration thus better print.

stevehooker, 2020-07-16:
M84 S300000 saved my life. Thank you. Was finally able to calibrate my extruder without it spitting out backwards, really fast, then trying to extrude 100mm slowly. I was comical.

DEFAULT_STEPPER_DEACTIVE_TIME = 0 doesn't seem to be working though. I'm still having my two Z-axis going wild and out of parallel. Will have to test some more. Still running calibration prints on this new build AM8.

I have some TMC2209s arriving soon, and will probably swap the two Z-axis and the extruder to them, Keeping the S42B on X and Y-axis. PIty because the prints I have at the moment seem very good.

stevehooker, 2020-07-19:
Yeah, I dumped the S42Bs and put on TMC2209, which arrived PDQ. All is much, much better, as my other printers.

I believe I was seeing moire patterns, salmon skin. I know Tampas from Nano Zero Stepper fame said that you can PID them out, but PID here is guesswork. I don't have the patience having lost so much time already.

I also tried balancing the Z drivers with the OLED display, such that the angle would be the some, or nearly the same, so they wouldn't get so crazily out of whack. No luck.

Gonna sell them on eBay.

ehtnevets, 2020-07-20:
Better spend time on Klipper. It really is a huge improvement over Marlin. The Pressure Advance and S-curve (real s-curve that you can hear from the noise the steppers make). S-curve is currently on the a different fork, https://github.com/dmbutyugin/klipper.

stevehooker, 2020-07-23:
I had Klipper on one 8bit machine. But upgraded it to 32bit and dropped it. Only saw Klipper as a workaround for 8bit boards.

Thank you for the **_quality_** heads-up. I'll look again ; -)

stevehooker, 2020-08-02:
For completeness. I found this which is for smart steppers:

"random, sporadic and inexplicable loss of positional accuracy which is caused by electrical noise"

https://www.estechnical.co.uk/signal-isolation-and-power-conditioning-for-smart-steppers