bigtreetech/BIGTREETECH-SKR-V1.3 #261: SKR v1.4 Turbo: can't get BLTouch working correctly

nakedandjameson, 2020-03-14:
Have SKR v1.4 Turbo working great on my Anycubic Chiron except for one thing; BLTouch. Have tried hooking it up to both Z-min endstop port and designated BLTouch ports with no difference in performance. BLTouch will self-test, deploy, and stow perfectly, but when I auto home or level the bed, the BLTouch just deploys and stows repeatedly before flashing the red light indicating an alarm.

Any ideas or solutions? Would appreciate any help. I can upload Marlin files if need be. Also have a video of the auto home sequence.

Also, tried M119 and it's always TRIGGERED for BLTouch. Tried inverting with no change in behavior, and tried changing sensor polarity with no change. Also tried swapping my BLTouch onto my friend's printer and it worked fine, so I know the BLTouch is okay.

nakedandjameson, 2020-03-17:
Note: M119 does not make the BLTouch do anything. During normal operation, M119 should trigger the BLTouch to deploy before reporting endstop positions, then stow afterward. This does not happen. BLTouch just stays stowed with red LED on (solid).

Sundancer78, 2020-04-01:
Hi,

you have to in the file
"\ Marlin \ src \ inc \ Conditionals_LCD.h"
change these lines like this:
[image]

in the configuration.h the line
Comment out "// # define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN" and
Define "#define Z_MIN_PROBE_PIN P0_10 // Pin 32 is the RAMPS default".

Gerggg, 2020-04-13:
the problem comes from a wire cut at the level of the skr
Ty for your help.