bigtreetech/BIGTREETECH-SKR-V1.3 #263: SKR 1.4 Endstop problem

Fuglys, 2020-03-17:
Hi guys, 

I'm using the SKR 1.4 Turbo with TMC2209 v3
So no matter what i do my enstop keeps telling me its open. 
using M119
Reporting endstop status
x_max: open
y_max: open
z_max: open
z_probe: TRIGGERED

My config.
Configuration.h
 #define USE_XMAX_PLUG
#define USE_YMAX_PLUG
#define USE_ZMAX_PLUG
 
// Enable pullup for all endstops to prevent a floating state
#define ENDSTOPPULLUPS
#if DISABLED(ENDSTOPPULLUPS)
  // Disable ENDSTOPPULLUPS to set pullups individually
  //#define ENDSTOPPULLUP_XMAX
  //#define ENDSTOPPULLUP_YMAX
  //#define ENDSTOPPULLUP_ZMAX
  //#define ENDSTOPPULLUP_XMIN
  //#define ENDSTOPPULLUP_YMIN
  //#define ENDSTOPPULLUP_ZMIN
  //#define ENDSTOPPULLUP_ZMIN_PROBE
#endif

// Enable pulldown for all endstops to prevent a floating state
//#define ENDSTOPPULLDOWNS
#if DISABLED(ENDSTOPPULLDOWNS)
  // Disable ENDSTOPPULLDOWNS to set pulldowns individually
  //#define ENDSTOPPULLDOWN_XMAX
  //#define ENDSTOPPULLDOWN_YMAX
  //#define ENDSTOPPULLDOWN_ZMAX
  //#define ENDSTOPPULLDOWN_XMIN
  //#define ENDSTOPPULLDOWN_YMIN
  //#define ENDSTOPPULLDOWN_ZMIN
  //#define ENDSTOPPULLDOWN_ZMIN_PROBE
 // Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
#define X_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define X_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Y_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MIN_PROBE_ENDSTOP_INVERTING false // Set to true to invert the logic of the probe.
 // Invert the stepper direction. Change (or reverse the motor connector) if an axis goes the wrong way.
#define INVERT_X_DIR true
#define INVERT_Y_DIR true
#define INVERT_Z_DIR false
 
// Direction of endstops when homing; 1=MAX, -1=MIN
// :[-1,1]
#define X_HOME_DIR 1
#define Y_HOME_DIR 1
#define Z_HOME_DIR 1
 // Min software endstops constrain movement within minimum coordinate bounds
#define MIN_SOFTWARE_ENDSTOPS
#if ENABLED(MIN_SOFTWARE_ENDSTOPS)
 #define MIN_SOFTWARE_ENDSTOP_X
 #define MIN_SOFTWARE_ENDSTOP_Y
 #define MIN_SOFTWARE_ENDSTOP_Z
#endif

// Max software endstops constrain movement within maximum coordinate bounds
#define MAX_SOFTWARE_ENDSTOPS
#if ENABLED(MAX_SOFTWARE_ENDSTOPS)
 #define MAX_SOFTWARE_ENDSTOP_X
 #define MAX_SOFTWARE_ENDSTOP_Y
 #define MAX_SOFTWARE_ENDSTOP_Z
#endif

Configuration_adv.h
No change for endstops. 

pins_BTT_SKR_V1_4.h 
Default 
 
//
// TMC StallGuard DIAG pins
//
#define X_DIAG_PIN         P1_29   // X-STOP
#define Y_DIAG_PIN         P1_28   // Y-STOP
#define Z_DIAG_PIN         P1_27   // Z-STOP
#define E0_DIAG_PIN        P1_26   // E0DET
#define E1_DIAG_PIN        P1_25   // E1DET

//
// Limit Switches
//
#if X_STALL_SENSITIVITY
 #define X_STOP_PIN       X_DIAG_PIN
 #if X_HOME_DIR < 0
 #define X_MAX_PIN      P1_26   // E0DET
  #else
 #define X_MIN_PIN      P1_26   // E0DET
  #endif
#else
 #define X_STOP_PIN       P1_29   // X-STOP
#endif

#if Y_STALL_SENSITIVITY
 #define Y_STOP_PIN       Y_DIAG_PIN
 #if Y_HOME_DIR < 0
 #define Y_MAX_PIN      P1_25   // E1DET
  #else
 #define Y_MIN_PIN      P1_25   // E1DET
  #endif
#else
 #define Y_STOP_PIN       P1_28   // Y-STOP
#endif

#if Z_STALL_SENSITIVITY
 #define Z_STOP_PIN       Z_DIAG_PIN
 #if Z_HOME_DIR < 0
 #define Z_MAX_PIN      P1_00   // PWRDET
  #else
 #define Z_MIN_PIN      P1_00   // PWRDET
  #endif
#else
 #ifndef Z_STOP_PIN
 #define Z_STOP_PIN     P1_27   // Z-STOP
  #endif
#endif

//
// Z Probe (when not Z_MIN_PIN)
//
#ifndef Z_MIN_PROBE_PIN
 #if Z_STOP_PIN != P1_27
 #define Z_MIN_PROBE_PIN P1_27
  #else
 #define Z_MIN_PROBE_PIN P0_10
  #endif
#endif

Showing a pic of my X end stop. 
Used to have another SKR 1.4 turbo and had everything working. however blow that one up with my thermostat. forgot to turn of printer while fiddling with it. 

Now i cant get even one endstop working. and i don't know why.  even tried changing pinout. to my E0DET pins. and still no luck. 

Am i missing something. 
hopefully you guys can help. 

Fugly

[image]

nakedandjameson, 2020-03-17:
I see you're homing to max positions rather than min positions for each axis; is that intentional?

Fuglys, 2020-03-18:
SKR 1.4 does not have min.

Talha909, 2020-03-21:
any luck.

Fuglys, 2020-03-21:
> any luck.

Nope. Even changes the pins to e0ed position. Still no success.

3D-McGee, 2020-03-21:
I think this is a case of where you have to remove a pin on the TMC2209.
I was having the same issue, couldnt get the Z end stop to read. until I did this.

[image]

https://github.com/bigtreetech/BIGTREETECH-TMC2209-V1.2/blob/master/TMC2209%20problems%20solving.pdf

Hope this helps you!

Fuglys, 2020-03-22:
> I think this is a case of where you have to remove a pin on the TMC2209.
> I was having the same issue, couldnt get the Z end stop to read. until I did this.
> 
> [image]
> 
> https://github.com/bigtreetech/BIGTREETECH-TMC2209-V1.2/blob/master/TMC2209%20problems%20solving.pdf
> 
> Hope this helps you!

I have tmc2209 v3. With these you don't have to remove the pins on x/y 
I had it working with my old skr 1.4 pro board until i blow it up with my temp sensor. It came lose and shorted out. 

So i used the same firmware but no matter what i do it always says open now. Even used new endstops from a old printer. 

Even changing the pins to another spot didn't work. Never had this problem with my old skr 1.4 pro. 

Also read this https://www.google.com/url?sa=t&source=web&rct=j&url=https://amp.reddit.com/r/BIGTREETECH/comments/ff8k2p/skr_14_tm2209_no_need_to_cut_diag_pin/&ved=2ahUKEwj48OqU3K3oAhWJPOwKHfq_D1IQFjAOegQICBAB&usg=AOvVaw2pCN2rf39jm9N8-0-wJPQV&cshid=1584867886877

Gonna check if i can make a vid + post my firmware somwhere. 

Anyway thx for the info. I will try and bent the pins just in case. Just to exclude the possibility

Fuglys, 2020-03-22:
> I have tmc2209 v3. With these you don't have to remove the pins on x/y
> I had it working with my old skr 1.4 pro board until i blow it up with my temp sensor. It came lose and shorted out.
>
> So i used [...]

So i did bent the pins. 
And it went from open to triggered.
However still no change when pressing endstop. I can see that endstop is working with the light that turns on on the endstop when pressing it. But its not registering. even tried reversing endstop its just keeps saying the same thing. Ether it stays triggered or open.

Reporting endstop status
x_max: open <- endstop not pressed
y_max: open
z_min: open
z_probe: TRIGGERED
>>> M119
SENDING:M119
Reporting endstop status
x_max: open <- endstop pressed
y_max: open
z_min: open
z_probe: TRIGGERED

I have a 2 pin endstop that i have put on GRD and 1.27 pins so 5v is clear.

Runningtarrens, 2020-05-26:
hi, maybe a bit late, but it seems that the SKR 1.4 has the same problem as the SKR1.3 , that there is a resitor soldered in where it shouldnt be. That means the z endstop pin doesnt work. you need to ether remove the resistor, which is really hard or asign another pin to the Z endstop and put the signal cable of your probe to this pin. I used the 2.00 pin on the SKR 1.3 and im going to use the 0.10 pin on the SKR 1.4.

i hope this helps

Psychokiller1888, 2020-08-02:
@Runningtarrens 
Unsolder and solder another resistor? Any link for instructions? I'm ok with unsoldering/soldering... Same problem on a skr 1.4 with optical endstops. Can't use mechanical for Z only, it's for a delta printer....

JRosello, 2020-08-02:
Check this videos maybe they can help....
Original
[](https://youtu.be/EivpEgsfNDc)
Update
[](https://youtu.be/gdecCW3vsIM)

robotsflo, 2020-08-10:
@Psychokiller1888 Did you solve this problem ?  i have receive mine skr 1.4 turbo 1 week ago and i have the same problem on X endstop ...

Runningtarrens, 2020-08-10:
> @Psychokiller1888 Did you solve this problem ? i have receive mine skr 1.4 turbo 1 week ago and i have the same problem on X endstop ...

this should be only a problem for the z endstop, not the x. did check the status and or switched the sensor to the y endstop just to test if it works?

Psychokiller1888, 2020-08-11:
Yes I did, by switching to E1DET, as PWRDET does not work for me, never triggers. But as said by @Runningtarrens this subject is only about Z endstop. X should be fine

nicobits46, 2020-09-17:
hey there i need some biiig help. i´ve been trying to seup the skr 1.4 turbo for over 3 months but i can´t Z and Y to home.
already cut  diag pin, endstops work perfect with M119
my printer is an anet a8 plus, skr 1.4 turbo with tmc2209 v1.2 BTT
stock display
I´ve tryed a bunch of configs but nothing seems to work.
Any ideas??
thank u

Fuglys, 2020-09-18:
> hey there i need some biiig help. i´ve been trying to seup the skr 1.4 turbo for over 3 months but i can´t Z and Y to home.
> already cut diag pin, endstops work perfect with M119
> my printer is an [...]

Show me your jumper settings below your tmc2209

nicobits46, 2020-09-22:
> Show me your jumper settings below your tmc2209

Hi I have them all in uart mode, 3rd row from black side left pin and middle pin
X works perfect, but I can't get Z and Y to home I have 2 skr 1.4 with the same problem

Fuglys, 2020-09-22:
[image]
Do you have your jumper like this. If not try this. Had loads of problem until i used this jumper settings. Had my doubts because of its cross section but it has been working for almost 6 months now

keimonde, 2021-02-03:
don't know about your build but googling around when setting up my turbo, I would try changing

#define Z_MIN_PROBE_ENDSTOP_INVERTING true

if your probe is servo type (bltouch&clonesmaybe) or npn inductive

nicobits46, 2021-02-03:
Awesome, thank you so much, I'll check that 💪🤘

lmatter, 2021-04-02:
me too, and watching.  I have a CR-10s and installed srk 1.4 turbo with 2209 drivers.  I have another printer (lotmaxx sc-10 shark)  with skr 1.4 and 2208s, it works great.  

Can't get any of the endstops to say anything but "open".  I am using USE_XMIN_PLUG instead of MAX.  Tried Z_MIN_PROBE_ENDSTOP_INVERTING  both ways.  Might be desperate enough to try the diagonal jumper trick posted above - currently set to UART mode.  Didn't bend any pins, hoping to avoid that.

nicobits46, 2021-04-02:
In my case I had to cut the pins and change the stock endstops in my Anet A8 plus for new ones.
They where set for NO and I needed to be NC

monstro-das-bolachas, 2021-04-03:
I upgraded my SKR 1.4 with 4x TMC2209 v1.2 and I'm running Marlin v2.0.6.1, from BTT's Github.
I cutted the Diag pins from the drivers for the Z axis and the Extruder stepper.
Since I made the upgrade and rewired everything, the Z inductive probe doesn't work.
I have a NPN NO 6-36V inductive probe with a voltage divider on the signal wire for a 5V signal. When triggered, I have 0.487V.
I have it connected to the GND of the probe connector (P0_10).
To troubleshoot, I temporarilly installed a mechanical switch and mimic the behavior, to see if the configuration was OK. Conclusion: it is ok! with the switch, works!

I tested the jumper configuration on the photo, with the jumper biases. Gives a message of TMC connection error and still doesn't work.

I measured the probe pins and I have around 3V. Could be a over voltage issue? Might be that the probe port doesn't tolerate 5V?

nicobits46, 2021-04-04:
Find me in telegram @nacc1981

monstro-das-bolachas, 2021-04-04:
Update:

just made a voltage divider to get around 3V and installed it on the signal cable.
After measuring the voltage, I have 3.8V when not triggered and 0.547V when triggered.
Still doesn't work.

Any idea what is going on?

lmatter, 2021-04-05:
> In my case I had to cut the pins and change the stock endstops in my Anet A8 plus for new ones.
> They where set for NO and I needed to be NCWhat brand are the drivers?

I finally bent the diag pins and now it detects TRIGGERED on the endstops.  (with USE_XMIN_PLUG, not MAX).  I was going to try sensorless homing but discovered that with a Z screw it won't work on that axis.

rudranshsinghmahra, 2024-10-10:
I have the updated fixed board of SKR 1.4 (old board has 101 register but new one has 01B Registor) but I am still unable to use optical end-stop in all the three axis. But the microswitch endstop works perfectly. I am using TMC 2209 by MakerBase stepper driver  Please let me know how can I fix it !!. Thanks