bigtreetech/BIGTREETECH-SKR-V1.3 #267: Your marlin 2.0 won't compile

gadgetman48, 2020-03-20:
your bigreetech marlin 2 doesn't compile  in atom or vscode,  "can't find build target"  It does compile only when  highlighting the platformio.ini in marlin 2 bugfix without errors.

RoscoeTheDog, 2020-03-25:
Can you elaborate how you got it to compile the other way? I am currently trying to compile it using the exact instructions from the PDF in the SKR 1.3 repository and I am also receiving #include errors which instantly fails building.

Everything has been configured as instructed. It seems as though it simply can't find the LPC1768 definition. I opened the project from the root folder of the Marlin firmware as instructed and triple checked the settings.

[image]

gadgetman48, 2020-03-25:
I am now getting it to compile from the root folders now but,
unfortunately, I don't know why.  In the worst of it, I had to highlight
Platformio.ini to get it to compile at all and that really doesn't make any
sense to me. I find Atom the worst IDE to work with.  It is slow, consumes
amazing amounts of resources and is capricious.   One thing i did was to go
back through configuration.h and configuration_adv.h and check everything
several times and found errors.  The board I am using is the turbo board
and I needed to use  BOARD_BIGTREE_SKR_V1_4_Turbo for the board name in
configuration.h  .

Michael B. Ryan MD

Home Of:
*EBlissPhotography.com* <http://EBlissPhotography.com>
EBliss Guitars <http://MBRyanGuitars.com>

EBliss Creativity LLC
home of: MBRyanGuitars <http://MBRyanguitars.com>EBliss  and EBliss
Photography LLC <http://EBlissPhotography.com>