bigtreetech/BIGTREETECH-SKR-V1.3 #272: Error: 'LPC176x' has not been declared

suckmo, 2020-03-24:
This is the same problem as ticket #130, however when attempting the proposed solution, the issue was not resolved.  The previous ticket identified a library version issue, specifically to rollback TMCstepper lib to v5.2. As you can see from the attachments the correct library dependency is being called.

[image]
[image]
[image]
[image]

Im attempting to compile Marlin Firmaware on windows 10 device with VScode. as per the instructions. I ve also attempted this on a linux based device only to end up at the same obstacle any advice would be greatly appricitated.

Sundancer78, 2020-03-28:
I also had this problem. I downloaded Marlin from the Marlin firmware website, unpacked the configuration files and everything into a new directory and started Visual Studio from there. From then on I had no problems compiling the firmware.

suckmo, 2020-03-31:
> I also had this problem. I downloaded Marlin from the Marlin firmware website, unpacked the configuration files and everything into a new directory and started Visual Studio from there. From then on I had no problems compiling the firmware.

Did you have still have to modify the TMC library to an older version?

Sundancer78, 2020-04-01:
I have left the current Library 0.6.2, I have no problems

rlmack1972, 2020-06-02:
My problem is a little different. I tried all the steps that everyone has tried and still get this:
[image]

rlmack1972, 2020-06-02:
Just to be clear I am using TMS2209 drivers and the motherboard is an MKS SGEN 1.0
This is the marlin bugfix 2.0 build

Here is a list of things that I have done:
set the configuration.h file settings
rolled back TMC library to 5.2
set platformio.ini info for TMC  TMCStepper@>=0.5.2,<0.6.0 in 3 locations

So I am out of ideas. Please help. If you need more info, just ask.