bigtreetech/BIGTREETECH-SKR-V1.3 #274: Skr 1.4 turbo does recognize card or usb

SkysoulPrt, 2020-03-25:
I have this problems in 2 boards that customized marlin.

set the initial configuration uploaded worked, connect only the steppers, check sensores & movement. disconnected, adjusted the configuration, uploaded with platformio,  the board stopped being recognized properly doesnt load the card, i did reset the firmware on sd card, but no luck, finaly i decided to plug the tft, a starts and locks with Err  max t E1, and thats it. Both of the board did the same with different configurations and just uploading directly with usb connected. So they both  boot and are and sometimes it is   recognized as marlin devices on pc .
[image]. Already tried to insert initial bin file on sd card but no success as well

Oatmeal15, 2020-03-26:
I have a similar issue and same boat with experience.

SkysoulPrt, 2020-03-30:
Solved, i searched the posts seems that skr 1.3 had the same problem, and ask on thread for files used so i put bin of 1.3, and suprisely it reflash and managed to get thing working from 0, and redo my config, so check this thread https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/223 
Thanks to 2  Sundancer78