bigtreetech/BIGTREETECH-SKR-V1.3 #275: Lags when printing via USB

angry-elf, 2020-03-26:
I have SKR 1.4 Turbo board.
I'm trying to print via USB port, connecting to Octoprint or desktop's Pronterface.

Looks like there are lags in connection. When I prints curved lines (like cylinders), there are small pauses causing oozes on this curves. Printing same gcode from SD (via TFT24) is ok, without pauses and oozes. 

Is it designed so? Can I use USB port on this board for printing? Is there specific options for marlin to resolve this issue?

kkosa1, 2020-03-26:
Had similar problems with SKR PRO V1.1 that were solved by updating Marlin to current bugfix.  It was a communications issue

angry-elf, 2020-03-26:
I tried to disconnect TFT24 leaving only Octoprint (via USB) connected and seems all fine now. Looks like this board (or Marlin) can't handle multiple hosts connected.

angry-elf, 2020-03-26:
Problem solved with new marlin build (2.0.5.2). Now TFT24 and Octoprint are connected, printing from Octoprint via USB is fine.

closed 2020-03-26 by angry-elf