bigtreetech/BIGTREETECH-SKR-V1.3 #278: UART mode on SKR V1.4 Turbo with Fystec TMC 2209 v3.0 [Solved]

emski0, 2020-03-29:
On the Fystec 2209 v3.0  the RX & TX pins are shifted compared to previous versions, so in order to use UART with SKR v1.4 board, you need to set the jumpers on SKR like the first picture, also you need to use the jumpers on the top of the 2209(like second picture). I found out this after a lot of fiddling and googling, this should be documented as it can result a lot of frustration for many people.
Please document this to help other people.

[image]
[image]

lucky62, 2020-04-05:
This solution may work, but shortening RX and TX on the driver is shortening 1K resistor on pin TX. I am not sure if this is important...
TX pin of FYSETC V3 is on the pin marked as SLP on A4988 and it is connected to the headers.
So better solution is to connect pins according picture. Then the 1K resistor is still in the effect.
[image]
Not also silkscreen error on FYSETC V3.0 driver:
[image]

lucky62, 2020-04-05:
[image]

emski0, 2020-04-06:
@lucky62 just to be clear, you don't have jumper on top of the tmc2209(Rx& TX) no?
Maybe your solution  is a better one, but mine is working for now, and i was looking for simplest solution, maybe i will give it a try later on and see if there is a difference

lucky62, 2020-04-06:
yes, no jumper on driver connecting RX&TX.

holstvoogd, 2020-04-25:
Just to confirm, both solutions work.
I haven't been able to figure out why the resistor on Tx is needed, but it's probably there for a purpose so I figured the CS-SLP wire is the better solution.
I've settled on soldering wires on the back of the board to keep it clean & reliable.

Fusseldieb, 2020-08-05:
> I haven't been able to figure out why the resistor on Tx is needed

Just chiming in, but I **guess** it's because the TMC chip could be 3.3V, while an Arduino has 5V logic, thus acting as a cheap solution to get it to safe voltages.